betaflight/betaflight #13602: Unnecessary features automatically enabled by default.

hydra, 2024-04-29:
### Describe the bug

After  issuing 'defaults' in the cli, and then 'diff' I get this:

```
feature SERVO_TILT
feature RANGEFINDER
feature DISPLAY
feature CHANNEL_FORWARDING
feature TRANSPONDER
```

The target has various, 'USE_...' defined, e.g. 'USE_SERVOS', but that shouldn't mean that features using the code enabled by the define should be used.

### To Reproduce

1) Build and flash any H7 target from source, tested on H7EXTREME and H7RF, likely happens on other targets too.  
2) issue 'defaults' in the cli.
3) wait for reboot.
4) go back to the cli and check the 'diff' output.

### Expected behavior

Only the features actually wanted are enabled.  i.e. none of the above should be enabled by default.

### Support ID

```txt
N/A
```

### Flight controller

Any H7

### Add any other context about the problem that you think might be relevant here

See `validateAndFixConfig`.

e.g.

```
#ifdef USE_RANGEFINDER
    featureEnableImmediate(FEATURE_RANGEFINDER);
#endif
#ifdef USE_SERVOS
    featureEnableImmediate(FEATURE_CHANNEL_FORWARDING);
    featureEnableImmediate(FEATURE_SERVO_TILT);
#endif
```

hydra, 2024-04-29:
This is tested on `4.5-maintenance` from yesterday's release of 4.5.0, commit: `c155f5830d0ffdee1c34071dd21f174ffc374c81`

ledvinap, 2024-04-29:
@hydra: someone spent considerable effort to enable these features (config and feature handling is total mess). 

Maybe we can use different values of USE_ define? 
```
#define FEATURE_DEFAULT_OFF 1
#define FEATURE_DEFAULT_ON 2
#define FEATURE_FORCE_ON 3

#define USE_SERVO FEATURE_DEFAULT_OFF 
```

with some improvement/rewording

hydra, 2024-05-14:
@ledvinap Not sure what the solution is, but there's some good discussion in https://github.com/betaflight/betaflight/pull/13608

hydra, 2024-05-19:
It seems there's some work going on in #13494 butwith reference to https://github.com/betaflight/betaflight/pull/13494/files#r1605977635 we must not forget the commit or additional PR that fixes the bug described in the OP https://github.com/betaflight/betaflight/issues/13602#issue-2268561318.

PR #13608 (2024-05-21): Do not enable features by default
- Fixes #13590
- Fixes #13602
- See also #13494 which should be merged first
- Remember #13244
closed 2024-05-21 by haslinghuis