betaflight/betaflight #13607: Mamba f435 2023A motors won't spin when arming

mlc76520, 2024-04-29:
### Describe the bug

Motors won't spin when arming. Same in BF motors tab; no spin. Works fine on 4.5.0 RC4.

### To Reproduce

Flash firmware. Spin motors in bf motors tab or arm kwad.

### Expected behavior

Motors should spin in bf motors tab. motors should spin when arming

### Support ID

```txt
Id: "daea838c-1b23-4a6d-b15b-a4076a3084ab"
```

### Flight controller

Mamba f435 2023A

### Other components

happymodel ep1 rx; DJI 03; Mambaf55A blheliS ESC / Bluejay flashed; vifly finder

### How are the different components wired up (including port information)

Elrs receiver on uart1
O3 on uart 3

### Add any other context about the problem that you think might be relevant here

https://github.com/betaflight/betaflight-configurator/issues/3921#issue-2269159021

haslinghuis, 2024-04-29:
Enable `set dshot_bidir = ON`

mlc76520, 2024-04-30:
Hi Mark - Thank you. Works fine now. 
But is it the intended behaviour since it was not mandatory on the previous RC4?

psfales, 2024-05-01:
Thank you.   My TMOTORF7 had the same issue:  Works with RC4, but motors don't spin in 4.5.0.   `set dshot_bidir=ON` got it working for me.   

Is this documented somewhere?  (Asking mainly because it seems like it could be impacting a lot of people.)

wterreb, 2024-05-04:
I encountered exactly the same issue.   Motors spin in  4.5.0 RC4,  but with the latest BF 4.5.0  the motors won't spin.  This is on a FLYWOOF745NANO flight controller.  I did not not try the set dshot_bidir=ON yet because I already reverted back to 4.5.0 RC4.

busheezy, 2024-05-10:
I had some help from discord. After explaining how I had to enable dshot_bidir, itsblunty let me know some people have had success setting dshot_bitbang to OFF. This worked for me and the motors now spin with dshot_bidir both ON and OFF.

alexberdnik, 2024-05-11:
Mamba f435 2023A
4.5.0 RC4 - motors are spinning with default configuration, don't work with dshot_bidir=ON.
4.5.0 - motors aren't spinning both with  dshot_bidir=ON and  dshot_bidir=OFF, BF configurator shows 100% motor errors. With bitbang=OFF motors are not detected.

spadeadam, 2024-05-13:
Same for me with Airbot F7 (F722) with 4.5.0 no motors and not sure what to switch on or off to get them turning….every previous version has worked ‘straight out of the box’ so have gone back to 4.4.
It seems a shame when all the developers have put so much time and effort into this latest release.

alexberdnik, 2024-05-22:
with 4.5.1-dev  motors are spinning with bidir dshot=off, not spinning when on. F435 2023a

alexberdnik, 2024-05-26:
I've also checked with two ESCs, both bluejay, J-H-15 and J-H-30, 48khz - bidirectional dshot doesn't work on these with the latest firmware. Original at32 firmware from product webpage (https://www.diatone.us/products/mamba-stack-mk5-f435-ai-icm42688p-flight-controller-55a-6s-8bit-esc-stack-30-5mm-m3) works.

victorscurs, 2024-06-30:
Had the same issue. Flashed it back to the OG 4.3 firmware, all is working

haslinghuis, 2024-06-30:
Did you check 4.5.1-dev ?

victorscurs, 2024-06-30:
> Did you check 4.5.1-dev ?

it wont compile in betaflight. couldnt try

haslinghuis, 2024-07-01:
MAMBAF435_2023A is not available as target.

closed 2024-07-01 by haslinghuis

alexberdnik, 2024-07-01:
Why it was dropped from dev branch?

haslinghuis, 2024-07-01:
https://github.com/betaflight/config/pull/426

shaneshort, 2024-11-16:
Has there been any update on this? Having a target pulled out from under us is a bit rough.

PaulGiek, 2025-01-08:
I can't even Find the Firmware anywhere. Which formware is approptiate for the Mamba f435 2023A? Thanks

victorscurs, 2025-01-08:
> I can't even Find the Firmware anywhere. Which formware is approptiate for the Mamba f435 2023A? Thanks

do you need the original one or updating the firmware?

nerdCopter, 2025-01-08:
from reading above and links, the problem is Diatone apparently released another hardware under the same name with conflicting resources.

that said, they (it) can be built. While these are unofficial, here are some 4.5.2 builds that may or may not contain the features you need.

[betaflight_4.5.2_AT32F435G_MAMBAF435_2023A_0d14c63cf.hex.zip](https://github.com/user-attachments/files/18351992/betaflight_4.5.2_AT32F435G_MAMBAF435_2023A_0d14c63cf.hex.zip) using original: https://raw.githubusercontent.com/betaflight/config/93dfc69de7bc909a6fd75a97954e38cfd59cd4d0/configs/MAMBAF435_2023A/config.h

[betaflight_4.5.2_AT32F435G_MAMBAF435_2023A2_0d14c63cf.hex.zip](https://github.com/user-attachments/files/18351995/betaflight_4.5.2_AT32F435G_MAMBAF435_2023A2_0d14c63cf.hex.zip) using config-PR # 426: https://raw.githubusercontent.com/x4FF3/betaflight-config/da27f4a023d0f19b61a99fd4cf77ca5304dde2f0/configs/MAMBAF435_2023A/config.h

via local files and `make MAMBAF435_2023A_rev MAMBAF435_2023A2_rev EXTRA_FLAGS="-DUSE_LED_STRIP -DUSE_GPS -DUSE_MAG -DUSE_DATA_READY_SIGNAL -DUSE_LED_STRIP"`

PaulGiek, 2025-01-09:
> > I can't even Find the Firmware anywhere. Which formware is approptiate for the Mamba f435 2023A? Thanks
> 
> 
> 
> do you need the original one or updating the firmware?

In general I can't even find anything but yes I'd like to update since I want to use usb_hid_cdc and it's not enabled in the original. Do you happen to know why there are no downloads on the Diatone website? Thanks

PaulGiek, 2025-01-09:
> from reading above and links, the problem is Diatone apparently released another hardware under the same name with conflicting resources.
>
>
>
> that said, they (it) can be built. While these are [...]

Thanks. That helps a lot!