betaflight/betaflight #13617: MAG causing flight controller freezing

ivrj, 2024-05-02:
### Describe the bug

When mag enabled, betaflight is crashing (freezing) on startup. If disable mag, flight controller wont freeze anymore.
Mag connected to CL2 DA2 as matek f405 se page orientation.
Hardware: Matek f405-se, gps Beitian BE-252Q  with magnetometer  QMC5883 , Caddix vista
Firmware BF 4.5

### To Reproduce

wiring hardware as describe on wiring session, enable magnetometer. 
If disable magnetometer, flight controller will not freeze anymore.
Need to put GPS powered on 5v to be able to configure on betaflight configurator, gps only turn on when connect LIPO, otherwise you can't change configuration if powered by USB (4.5v line).

### Expected behavior

flight controller not freezes 

### Support ID

```txt
Id: "9f31f0fe-4d28-4b24-a7e6-e71ff2878534"
```

### Flight controller

Matek F405 SE, Aliexpress

### Other components

GPS Beitian BE-252q  with mag QMC5883
Caddix Vista
Crossfire nano

### How are the different components wired up (including port information)

1. Caddix vista on UART5  powered on VBat
2. GPS on UART4  powered on 5v line 
3. Crossfire on UART2 powered on 4.5v
4. Mag QMC5883 on CL2/DA2  as  matek product page orientation ( https://www.mateksys.com/?portfolio=f405-se#tab-id-4 and https://www.mateksys.com/?portfolio=f405-se#tab-id-5) 👇

[image]

[image]

ivrj, 2024-06-02:
issue still there.

haslinghuis, 2024-06-02:
Did you check with 4.6 if issue persists?

alexrjohnston, 2024-07-09:
I have the same issue with BF4.5 and Matek F405SE + mag - freezing. with 4.6 I can't even get into the configurator - lights on the FC init then I get one blue flash then its frozen to single solid red light

ctzsnooze, 2024-09-05:
The target file for that board lists two i2c channels, with baro on 1 and mag on 2.
I'm not sure how the i2c configuration should be set, but the target file may not have it correctly done.

Check in CLI, with `get MAG`, and look at the `mag_i2c_device` setting.  The `mag_i2c_address` should be zero, for auto-assignment of the address on that i2c 'device' or 'bus'.

Also check with `get BARO` to see what it's settings are.

You may try some experimentation... possibilities include:
- disable MAG and check that BARO works with the current settings.
- disable BARO and see if MAG now works.
- try using a different value (0,1,2) for the `mag_i2c_device` with BARO disabled.
- try using a different value (0,1,2) for the `baro_i2c_device` with MAG enabled.

It will be some annoying interaction between the i2c settings.

closed 2026-09-26 by blckmn