betaflight/betaflight #13618: Neutron F435 AIO no DSHOT output

error414, 2024-05-02:
### Describe the bug

Neutron F435 AIO with AM32 ESCs. There is not DSHOT outpout for BF 4.5-RC4+. Tested in motor tab.

[4.5.0-RC1](https://github.com/betaflight/betaflight/releases/tag/4.5.0-RC1) WORKS
[4.5.0-RC2](https://github.com/betaflight/betaflight/releases/tag/4.5.0-RC2) WORKS
[4.5.0-RC3](https://github.com/betaflight/betaflight/releases/tag/4.5.0-RC3) WORKS
[4.5.0-RC4](https://github.com/betaflight/betaflight/releases/tag/4.5.0-RC4) NO DSHOT output, motor doesn not spin (works if bidirection dshot is ON)
[4.5.0](https://github.com/betaflight/betaflight/releases/tag/4.5.0) NO DSHOT output, motor doesn not spin (works if bidirection dshot is ON)

### To Reproduce

Flash 4.5.0 into neutron F435mini, go to motor tab, test motors

### Expected behavior

dshot output

### Support ID

```txt
*** Support data submitted *** 
Id: "2c924fd6-2c2b-4ed9-81cd-ec2013125619"
```

### Flight controller

Neutron F435 AIO AM32

### Other components

xing 1404 4600kv

### How are the different components wired up (including port information)

no other component connected

### Add any other context about the problem that you think might be relevant here

MCU AT32F435 Clock=288MHz, Vref=3.27V, Core temp=25degC
Stack size: 2048, Stack address: 0x2002fff0
Configuration: CONFIGURED, size: 3771, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: MSP (53 x 20)
BUILD KEY: b817110c6bae6144a0c573448561c7c4 (4.5.0)
System Uptime: 9 seconds, Current Time: 2024-05-02T19:56:06.837+00:00
CPU:8%, cycle time: 314, GYRO rate: 3184, RX rate: 15, System rate: 9
Voltage: 4 * 0.01V (0S battery - INIT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP

iHexile, 2024-05-03:
Having the same issue with my JHEF7DUAL fc, was working fine and then flashed BF 4.5, BLHeli configurator reads from the esc via passthrough, but no spinning when trying to arm or use BF motor direction configurator

Sekilsgs2, 2024-05-04:
Same here.. FC BayacRC Dolphin. 

And some errors with dshot600l, with dshot300 - no errors.
And RPM filtering not working normal - create some resonance - without RPM filtering no problems..

MukaroFPV, 2024-05-10:
Having similar issues with DShot 600 and 300 as well. I'm running a Tmotor VeloxF7SE. Flashed betaflight 4.5 on it and when I test motors erros on dshot telemetry appear, usually just in one motor. And it reacts when other motors are spinning. Flashed back t9 4.4 and it's fine not errors. Bug in Dshot??

alexberdnik, 2024-05-19:
which version of AM32 do you have installed on NeutronRC F435? Mine (35A version) works fine with 4.5.

error414, 2024-05-21:
I use ESC 1: NEUTRON - AM32, 2.0, Bootloader v1 (PB4), MCU: F421

alexberdnik, 2024-05-21:
Can you try with 2.10 (available here: https://github.com/am32-firmware/AM32/releases/tag/v2.10) ?

haslinghuis, 2024-05-21:
Perhaps also try 4.5.1-dev (first)

error414, 2024-05-21:
4.5.1-dev works

haslinghuis, 2024-05-21:
Fixed in [2d3c8ee](https://github.com/betaflight/betaflight/commit/2d3c8eea334752ca085cdaf3c7bb52996c3ef28c)

closed 2024-05-23 by haslinghuis