betaflight/betaflight #13619: Blackbox logging on CLRACING F4 (JB-Board) up from >4.4.x not possible

mrRobot62, 2024-05-03:
### Describe the bug

# BF 4.4.x (Blackbox recording not possible)
I tested FWversion up from 4.4 to 4.5 and every time I struggle with this error
[image]

# BF 4.3.2 (Blackbox recording OK)
with BF 4.3.2 with this board it is possible to activate BB logging.
[image]

Both firmwares tested this with **BF-Configurator 10.10.0**

### To Reproduce

Install 4.3.2 and check if Blackbox logging is possible like above image.
Change to >= 4.4.3 and you will get the error like in above image. I tried it with 4.5.0 with the same issue

### Expected behavior

**Expected behaviour:**
BB logging should be possible with >= 4.4.x FW version.

### Support ID

```txt
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.31V, Core temp=40degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked
GYRO=MPU6000, ACC=MPU6000
OSD: MSP
System Uptime: 227 seconds, Current Time: 2024-05-03T10:02:15.319+00:00
CPU:69%, cycle time: 128, GYRO rate: 7812, RX rate: 15, System rate: 9
Voltage: 20 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216
Paritions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump master

# version
# Betaflight / STM32F405 (S405) 4.3.2 Nov 28 2022 / 07:26:30 (60c9521) MSP API: 1.44

# config: YES

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id CLRA

# name: URUAV

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
[... 1178 lines cut ...]
    CH1 : CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

### Flight controller

CLRacing F405, Joshua Bardwel brand

### How are the different components wired up (including port information)

UART 4 = CRSF
UART 6 = Walksnail VTX
nothing more, no LEDs , no beeper

[image]

### Add any other context about the problem that you think might be relevant here

I tried some cli commands around sd-card without any success. 
I tried the same sd-card config in cli from 4.3.2 and tried them in 4.4.3 and 4.5.0 without success.

Additional notes:
following manual, I have to setup SDCARD for BB-logging but there is NO SDCARD (physically) used

haslinghuis, 2024-05-03:
@mrRobot62 please check https://github.com/betaflight/unified-targets/issues/1008

mrRobot62, 2024-05-03:
Currently no success - tried all Flash chip which are defined, flashed 4.5.0 again - no success to. Did it based on #925 & #1008

mhm

mrRobot62, 2024-05-03:
SUCCESSFULLY 

The solution is to set
EXPERT MODE (as advised)
in CUSTOM DEFINES
`FLASH FLASH_W25Q128FV`

Without `FLASH` it will not work on this FC

haslinghuis, 2024-05-03:
@mrRobot62 did you try with CLRACINGF4V2 as that config is the same - only swapping SDCard for flash chip

```bash
#define USE_FLASH
#define USE_FLASH_W25Q128FV
```

mrRobot62, 2024-05-03:
ah - no I didn't try this, because I didn't that the diff between both configs are the SDCard / Flash chip things.

However, thanks for your brain kick ;-) - problem solved know for CLRACINGF4 - and maybe other owners of this FC can participate from this thread

closed 2024-05-03 by mrRobot62