betaflight/betaflight #13622: MSP OSD Element "GPS Sats" not showing up

x8n1620, 2024-05-04:
### Describe the bug

I've search for the issue, found  #13484 , but that is not the same issue

After Upgrading from 4.4.3 to 4.5 the OSD Element "GPS Sats" ist gone. No matter if osd_gps_sats_show_pdop is set to ON or OFF.

All other of my OSD-Elements works fine, including the GPS Position, Speed, Altitude.... Also RTH working fine.
Only the OSD Element is gone (was available with 4.4.3 and older).

Setup:
FC: Holybro Kakute H7
VTX: Caddx Vista with WTFOS
DJI Goggles v2 with WTFOS

### To Reproduce

Flashing a Holybro Kakute H7 to BF 4.5
Wire up an Vista with WTFOS and MSP Display
Connect to DJI Googles v2

### Expected behavior

OSD Elemen GPS Sats is showing up in OSD

### Support ID

```txt
*** Support data submitted *** 
Id: "35e5d308-b05e-437b-b189-765b79a45628"
```

### Flight controller

Holybro Kakute H7 

### Other components

VTX: Caddx Vista with WTFOS
DJI Goggles v2 with WTFOS
HGLRC M100 Mini GPS 10

ordningsproblem, 2024-05-06:
This is a known issue with HDFonts in BF 4.5.  
Workaround is to 'set displayport_msp_fonts = 0,0,0,0'

x8n1620, 2024-05-08:
Works fine, thank your!

closed 2024-05-08 by x8n1620