betaflight/betaflight #13628: Motors won't spin in configurator with bidirectional dshot disabled

freasy, 2024-05-07:
### Describe the bug

When you have bidirectional dshot disabled, it won't spin the motors through the configurator.

### To Reproduce

Setup a flight controller + ESC.
Set dshot(300 or 600)
Disable bidirectional dshot
Try to spin through motors tab in betaflight configurator 

### Expected behavior

It will spin motors.

### Support ID

```txt
0
```

### Flight controller

Speedybee V4 F722

VincentHuang21, 2024-05-10:
I have the same issue

Zorbie555, 2024-05-15:
I have the same problem, with RPM filter motor 1 and 2 does not spin, without RPM filter neither spins. With RC4, at least it worked without RPM filter.  Greatmontain AT32F435 FC, iFlight succex 45a BLHELI_S ESC.

closed 2024-05-17 by github-actions[bot]