betaflight/betaflight #13638: TMOTOR Velox F7 SE flight controller has high gyroscope noise

dengdizzy, 2024-05-14:
### Describe the bug

The flight controller using 4.5.0 firmware has high noise with gyro

### To Reproduce

1.cloud build 4.5.0 TMTR-TMOTORVELOXF7SE target,and flash to FC
2.start to fly
3.into cli and enter:
```
resource MOTOR  5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE

resource MOTOR 5 C08
resource MOTOR 6 C09
resource MOTOR 7 A08
resource MOTOR 8 A09

save
```

4.start to fly

### Expected behavior

Two flight will have same gyro noisy,but actually there are dirffrent.

### Support ID

```txt
1a16c0ee-2423-493e-8ce7-0c1cd11f092b
```

### Flight controller

TMOTOR Velox F7 SE

### Other components

RX:ELRS 3.0 receiever
VTX:analog
ESC:V50A SE

### How are the different components wired up (including port information)

RX:UART3
VTX:UART2
ESC:MOTOR1-MOTOR4

dengdizzy, 2024-05-16:
@sugaarK Can you find the reason?

sugaarK, 2024-05-16:
Absolutely no idea!  Lacking in details or logs. Noise could be from esc’s, frame, build, props, bad tune 🤔

SupaflyFPV, 2024-05-16:
> ### Describe the bug
> The flight controller using 4.5.0 firmware has high noise with gyro
> 
> ### To Reproduce
> 1.cloud build 4.5.0 TMTR-TMOTORVELOXF7SE target,and flash to FC 2.start to fly 3.into cli and enter:
> 
> ```
> resource MOTOR  5 NONE
> resource MOTOR 6 NONE
> resource MOTOR 7 NONE
> resource MOTOR 8 NONE
> 
> resource MOTOR 5 C08
> resource MOTOR 6 C09
> resource MOTOR 7 A08
> resource MOTOR 8 A09
> 
> save
> ```
> 
> 4.start to fly
> 
> ### Expected behavior
> Two flight will have same gyro noisy,but actually there are dirffrent.
> 
> ### Support ID
> ```
> 1a16c0ee-2423-493e-8ce7-0c1cd11f092b
> ```
> 
> ### Flight controller
> TMOTOR Velox F7 SE
> 
> ### Other components
> RX:ELRS 3.0 receiever VTX:analog ESC:V50A SE
> 
> ### How are the different components wired up (including port information)
> RX:UART3 VTX:UART2 ESC:MOTOR1-MOTOR4
> 
> ### Add any other context about the problem that you think might be relevant here
> _No response_

please can you provide a log and a backup of the setup (from preset page)

dengdizzy, 2024-05-17:
Strange,we use cloud build today,this time become normal.So if next time we meet same situation,i will open this issue.

closed 2024-05-17 by dengdizzy