betaflight/betaflight #13640: Cannot change GPS Provider once set as MSP

wrightwells, 2024-05-15:
### Describe the bug

Changed my GPS to MSP and now it will not change to anything else and device shows as rebooting on the GUI and will not reconnect. unplugging the usb to reconnect then shows that the GPS Priovider has remained as MSP.

### To Reproduce

GPS Page GPS Configuration section
Set Protocol in GUI to MSP, Save and Reboot

Go back to GUI and change to UBLOX or NMEA, Save and Reboot
or in CLI set gps_provider=UBLOX then save 

### Expected behavior

The GPS Provider should change to UBLOX or NMEA

### Configurator version

10.10.0 (c97deaf)

### Flight controller configuration

```txt
# version
# Betaflight / STM32F405 (S405) 4.5.0 Apr 28 2024 / 02:46:01 (c155f5830) MSP API: 1.46
# config rev: 3068e6e

# start the command batch
batch start

board_name SPEEDYBEEF405MINI
manufacturer_id SPBE

# name: AOS 3.5

# feature
feature -AIRMODE
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD
feature ESC_SENSOR

# beacon
beacon RX_SET

# aux
aux 0 0 0 1500 2100 0 0
aux 1 2 1 900 1300 0 0
aux 2 13 6 1500 2100 0 0
aux 3 28 1 1300 2100 0 0
aux 4 35 4 1500 2100 0 0
aux 5 36 5 1500 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 29 1 0 0
adjrange 1 0 3 900 2100 12 3 0 0

# rxfail
rxfail 3 s 1400
rxfail 6 s 2000
rxfail 10 s 2000

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_min_hz = 150
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_trim_pitch = -1
set acc_trim_roll = -1
set acc_calibration = 41,95,-9,1
set min_check = 1000
set max_check = 2000
set dshot_idle_value = 300
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = MSP
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_tim_1_pos = 7527
set osd_tim_2_pos = 7559
set osd_flymode_pos = 6231
set osd_craft_name_pos = 6165
set osd_gps_lon_pos = 7202
set osd_gps_lat_pos = 7234
set osd_gps_sats_pos = 7266
set osd_home_dir_pos = 6329
set osd_home_dist_pos = 6296
set osd_altitude_pos = 7298
set osd_avg_cell_voltage_pos = 6550
set rpm_filter_weights = 100,0,80
set craft_name = AOS 3.5
set pilot_name = 

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 80
set dterm_lpf1_dyn_max_hz = 110
set dterm_lpf1_type = BIQUAD
set dterm_lpf2_static_hz = 0
set vbat_sag_compensation = 100
set p_pitch = 57
set i_pitch = 102
set d_pitch = 57
set f_pitch = 172
set p_roll = 64
set i_roll = 115
set d_roll = 50
set f_roll = 195
set p_yaw = 64
set i_yaw = 115
set f_yaw = 195
set d_min_roll = 37
set d_min_pitch = 42
set d_max_advance = 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 6
set feedforward_max_rate_limit = 95
set dyn_idle_min_rpm = 45
set dyn_idle_p_gain = 40
set simplified_master_multiplier = 125
set simplified_pi_gain = 115
set simplified_feedforward_gain = 130
set simplified_pitch_pi_gain = 85
set simplified_dterm_filter = OFF
set ez_landing_limit = 5

rateprofile 1

# rateprofile 1
set thr_mid = 25
set thr_expo = 25
set roll_rc_rate = 4
set pitch_rc_rate = 4
set roll_srate = 40
set pitch_srate = 40
set yaw_srate = 69
set throttle_limit_type = SCALE
set throttle_limit_percent = 60

# end the command batch
batch end
```

Onizuka666777, 2024-09-16:
> remove MSP from ports tab for the uart port

how remove MSP from ports tab for the uart port?

Onizuka666777, 2024-09-18:
> Remove the GPS from the UART that it is being used by in Ports tab <img alt="Screenshot 2024-09-16 at 15 20 56" width="1828" src="https://private-user-images.githubusercontent.com/21678359/367798922-23b94238-7098-486f-a4be-2e4de06d81a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2NDM2MzUsIm5iZiI6MTcyNjY0MzMzNSwicGF0aCI6Ii8yMTY3ODM1OS8zNjc3OTg5MjItMjNiOTQyMzgtNzA5OC00ODZmLWE0YmUtMmU0ZGUwNmQ4MWEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDA3MDg1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVmMWVjZDlmMjkwNWYzZDNkYWNlZmY5M2FhM2Y1Y2U5ZWUyZmY5ZGZmY2I3MDE2MzU0YmQ4NDVlODIxZTcxMDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6oqABVAOdz2V3KDw12sGFGM-QLEreeNhFjJKqjt81fI">
> 
> then in CLI set gps_provider = UBLOX

okey, i try..
thanks bro...

kompoot, 2025-02-23:
removed the GPS from uart6 and rebooted

[image]

then i went to cli and "set gps_provider" to verify it's wrong which it is

[image]

so "set gps_provider = UBLOX"

[image]

then i enter exit to close the cli, cli closes and disconnects fc from betaflight.
i connect to betaflight again go to cli to double check if it's now on UBLOX "set gps_provider"
aaaand it's back to MSP

[image]

anyone knows how to fix?

haslinghuis, 2025-02-23:
save after making changes

kompoot, 2025-02-23:
I thought "save to file" saves it didn't know you can just type i "save" in cli, thank you

nerdCopter, 2025-02-24:
typing `save` is the only way to save anything set via CLI.

bennie1995, 2025-02-28:
> feature -gps
> save
>
> set gps_provider = ublox
> feature gps
> save
>
> Setup gps in ports tab

Thanks, but again done all this and it’s just won’t save, then it doesn’t
accept any cli dumps, once I unplug the fc and put it back in I’m back to
the same thing lol

closed 2026-09-26 by blckmn