betaflight/betaflight #13643: Allow modes to link to ARM condition

psfales, 2024-05-16:
### Is your feature request related to a problem? Please describe

I would like to have a user mode (specifically LED running lights) which are controlled by a switch, but also only active when ARMed.

Even though "ARM" is a mode on the modes tab, it doesn't show up as a choice for "Add Link"

### Describe the solution you'd like

Add ARM to the list of modes in the dropdown menu for "Add LInk"

### Describe alternatives you've considered

I could use "Add Range" and add a range for the same AUX1 channel that controls arming.   However, that's not quite the same thing because that would activate the mode whenever arming is *requested* even if it can't arm for some reason.    I would prefer to have the lights come only when quad is actually armed and ready to fly.

psfales, 2024-06-01:
Ran into another use case for this... Wanted to enable blackbox only when the quad is Armed, but it's not possible to link to that mode.   Again, the short term workaround is to use a range that is activated by the arm switch, but that turns on recording when the switch is on, even if the quad does not actually arm.

Quick-Flash, 2024-06-01:
> Ran into another use case for this... Wanted to enable blackbox only when the quad is Armed, but it's not possible to link to that mode. Again, the short term workaround is to use a range that is [...]

All of this is currently possible based on how you setup your switches. Its possible to link your arming switch to also be a prerequisite for other switch modes to be enabled. Its not a feature that is widely used, but it is currently possible.

psfales, 2024-07-02:
I realize that this is a feature request, not a bug report, but I'm hoping it can continue to be considered as a future enhancement.

closed 2024-08-10 by github-actions[bot]