betaflight/betaflight #13645: F722 cannot turn on the led strip and GPS

YangChen1234, 2024-05-17:
### Describe the bug

First of all, my flight control is AOCODARCF722MINIV2. Use‘’ https://github.com/betaflight/betaflight/releases ‘’to download betaflight_4. 5.0 _STM32F7X2  Hex  and betaflight_4. 4 . 2 _STM32F7X2  Hex  firmware to import the configuration file,  use the version is 10.10.0 ground station.,found that GPS and LED strips function could not be turned on.
[image]

### To Reproduce

But I downloaded the firmware at the BF ground station and imported the same configuration file to turn on the GPS and led strips 
![Uploading QQ浏览器截图
[image]
20240517152831.png…]()

### Expected behavior

https://github.com/betaflight/betaflight/assets/89179705/20f87b56-04a1-400e-8776-9a187bfe049a

### Support ID

```txt
###

# # Problem description

# # support id

# ###

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Apr 28 2024 / 01:55:47 (c155f58) MSP API: 1.46
# board: manufacturer_id: SJET, board_name: AOCODAF722MINIV2

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=60degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3876, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6500, ACC=MPU6500, BARO=DPS310
OSD: MSP (53 x 20)
System Uptime: 20 seconds, Current Time: 2024-05-17T07:47:16.045+00:00
CPU:30%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 4 * 0.01V (0S battery - INIT)
I2C Errors: 11
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS ANGLE CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump master

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Apr 28 2024 / 01:55:47 (c155f58) MSP API: 1.46

# start the command batch
batch start

board_name AOCODAF722MINIV2
manufacturer_id SJET

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
[... 1007 lines cut ...]

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

#
```

### Flight controller

AOCODARC F722MINIV2

lida2003, 2024-05-17:
Did you try 10.9.0 configurator?

haslinghuis, 2024-05-18:
Use cloud of build locally with features selected.

closed 2024-05-18 by haslinghuis

YangChen1234, 2024-05-24:
> Use cloud of build locally with features selected.
Thank you for your reply.I want to locally compile 4.4.3 firmware on Ubuntu to add GPS and LEDs function how to operate

[image]