betaflight/betaflight #13647: Ublox M10 - Receiver (Dronetag BS) - Disabled Autoconfig

optical-o, 2024-05-20:
### Describe the bug

The issue appeared on version 4.5.0 but was not present on version 4.4.3. The issue is that Betaflight is receiving the GNSS UBX messages PVT, DOP, and STATUS; however, the drone does not appear to validate the presence of the GNSS Receiver, as shown in the attached photo from the Betaflight configurator.

[image]

The problem can be resolved by disconnecting the GNSS receiver(Dronetag BS), checking the Auto-config option on, saving and Rebooting the Drone, setting the Auto-config option to off, saving and Rebooting the Drone, and connecting the GNSS receiver. The beta flight then successfully recognizes the GNSS receiver. (Note that the receiver is disconnected during this operation, so no configuration change has occurred on the Receiver side.) 

[image]

Here is a UBX capture of the data sent to the beta flight controller by the GNSS receiver for validation. I'm happy to talk to you about any requirements we may be missing on the side of the Dronetag BS firmware. However, since changing the configuration and rebooting the drone somehow solves the problem, there might be a deeper issue inside the beta flight firmware.

[ubx_log.ubx.zip](https://github.com/betaflight/betaflight/files/15377544/ubx_log.ubx.zip)

Additional question:
Is there some other messages that we are missing that do you require than: 
PVT(10Hz), DOP(10Hz), SAT(1Hz), and STATUS(0.0333Hz (30s))
To verify that everything should work as expected. As attached ubx log file.

### To Reproduce

The description already contains the steps to reproduce as part of the description.

### Expected behavior

Bettaflight should recognize the GNSS receiver right after booting.

### Support ID

```txt
*** Support data submitted *** 
Id: "b8a62d6c-e22c-4436-83f7-958e6b773abd"
```

### Flight controller

4.5.0 BTFL

optical-o, 2024-06-05:
Is there any way I can contribute to this issue from my side? Maybe I can have a quick call with any of the developers?

McGiverGim, 2024-06-13:
I don't understand exactly when and when not the GPS works. The support ID provided has this two parameters:
```
set gps_auto_config = OFF
set gps_auto_baud = OFF
```
Maybe it corresponds at the second situation. Can you try to enable BOTH of them to see if the GPS is detected and configured correctly?

optical-o, 2024-06-17:
I have tested the configuration with both of the settings ON.

set gps_auto_config = ON # Works bot in the UI, it always shows 57600, which is incorrect.
set gps_auto_baud = ON

[image]

The result is the same: Betaflight receives messages with information but does not confirm the presence of the GNSS receiver. However, I could not reproduce my result from the previous testing, which showed that the beta flight with some restarts detects the GNSS receiver. The beta flight remained unconnected to the GPS receiver and did not show the Satelite info even though it was receiving the NAV-PVT messages obvious from the image. 

Can you look at what needs to be received for the beta flight to mark the receiver as Connected? I will then analyze the communication and whether the messages are getting to the flight controller.

Note: The Dronetag BS is configuring the M10 Reciever, which passes through the I2C interface to the UART interface. Some specific configurations regarding the UART port might not be reflected on the I2C. 

```
status result for GNSS
GPS: NOT CONNECTED, UART8 19200 (set to 115200), NOT CONFIGURED, version =  unknown
```

optical-o, 2024-08-12:
I  have traced the issue to Baud detection inside the betaflight. 
Why does the beta flight auto bauds always, even when a predefined baud rate is configured for the port ?
Why was the option to disable the auto bauding removed from the betaglight 4.5.0 ?

@McGiverGim

ledvinap, 2024-08-12:
@optical-o : Can you post (Saleae) trace of GPS communication?

From code, it seems that BF is waiting for version reply (`ubloxSendClassMessage(CLASS_MON, MSG_MON_VER, 0);`). Only these versions are supported: https://github.com/betaflight/betaflight/blob/master/src/main/io/gps.c#L472

(BTW: můžete mne kontaktovat na gmailu nebo telefonu)

optical-o, 2024-08-12:
Hello @ledvinap 
I have identified that the issue is the TX  communication into the Dronetag BS. The Flight controller is for some reason unable to drive the Output when configured as UART port. Otherwise it has solid 1.8V. What happends is that the Dronetag BS does not recieve the UBX Poll for the HW MON so the betaflight thinks that the GNNS receiver is not there. Would it be possible so that the GPS skips the autobauding when baudrate is preconfigured by the user ? 

I'm testing with:
https://newbeedrone.com/products/newbeedrone-infinity-aio-v2-fc-esc-combo

However, we have users with variouse hardware that reports same issues.

We will release an firmware update that will enable pull up on the RX of the dronetag BS that will solve this issue.
Do you have some information about the drive strength of the FC for the UART ? Generaly what are the volatges of the Flight controlles for the UART interfaces ?

ledvinap, 2024-08-12:
@SteveCEvans : https://github.com/betaflight/betaflight/pull/13021 is probably causing this problem. TX is disabled when pin is Low (in reset/default state, so without pullup).

PR #14052 (2024-11-30): Workaround: SERIAL_CHECK_TX is broken on F7
SERIAL_CHECK_TX STM32F7 does not configure TX pin as pullup before checking it. This prevents transmission when pin is not pulled up (#13647, #14050). If you need SERIAL_CHECK_TX, define USE_F7_CHECK_TX in build.

This can be reverted when F7 SERIAL_CHECK_TX is fixed.

Fixes #13647, #14050
closed 2024-11-30 by blckmn