betaflight/betaflight #13650: Allow "user friendly" names for user modes

psfales, 2024-05-22:
### Is your feature request related to a problem? Please describe

I'm always frustrated when looking at the modes tab, and can't remember what I was using "User 1" for

### Describe the solution you'd like

Allow the user to enter a meaningful name like "Running Lights" for a user mode instead of the default name of "User 1"

### Describe alternatives you've considered

This is definitely a frill, as it's functional the way it is.  But I think it would be a nice improvement!

haslinghuis, 2024-05-22:
You did not find the Easter egg

[image]

closed 2024-05-22 by haslinghuis

psfales, 2024-05-22:
Oops.  My face is red!  Thanks for setting me straight.

In my defense, I don't think this is widely known.   (I've been reading the on-line tutorials about configuring and using the user modes, but I don't think anyone talked about naming them!)

Thanks again.  I guess it was pretty easy to implement that feature request!