betaflight/betaflight #13652: Occasionally wrong rssi values when ELRS is not connected

catkira, 2024-05-24:
### Describe the bug

When no ELRS connection is present, I get occasionally wrong RSSI values. I would expect only 0 values, but sometimes I get 1022.

timestamp,voltage,amperage,mAhdrawn,rssi
1716542447264,20.87,0.15,1,0
1716542447359,20.87,0.15,1,0
1716542447461,20.87,0.15,1,0
1716542447554,20.87,0.15,1,1022
1716542447664,20.87,0.15,1,1022
1716542447769,20.87,0.15,1,1022
1716542447864,20.87,0.15,1,1022
1716542447959,20.87,0.15,1,0
1716542448054,20.87,0.15,1,0
1716542448165,20.87,0.15,1,0
1716542448264,20.87,0.15,1,0
1716542448365,20.87,0.15,1,0
1716542448460,20.87,0.15,1,0
1716542448559,20.87,0.15,1,0
1716542448662,20.87,0.15,1,0
1716542448774,20.87,0.15,1,0
1716542448868,20.87,0.15,1,0
1716542448964,20.87,0.15,1,0
1716542449062,20.86,0.15,1,0
1716542449163,20.85,0.15,1,0
1716542449264,20.85,0.15,1,0
1716542449358,20.86,0.15,1,0
1716542449462,20.85,0.15,1,0
1716542449561,20.86,0.15,1,0
1716542449660,20.86,0.15,1,0
1716542449762,20.86,0.15,1,0
1716542449861,20.86,0.15,1,0
1716542449961,20.85,0.15,1,0
1716542450057,20.85,0.15,1,0
1716542450164,20.85,0.15,1,0
1716542450263,20.85,0.15,1,0
1716542450364,20.85,0.15,1,0
1716542450459,20.85,0.15,1,0
1716542450565,20.85,0.15,1,0
1716542450662,20.85,0.15,1,0
1716542450773,20.85,0.15,1,0
1716542450881,20.85,0.15,1,0
1716542450960,20.85,0.15,1,0
1716542451055,20.87,0.15,1,0
1716542451159,20.87,0.15,1,0
1716542451262,20.87,0.15,1,1022
1716542451373,20.87,0.15,1,1022
1716542451468,20.87,0.15,1,0
1716542451556,20.87,0.15,1,0
1716542451661,20.87,0.15,1,0
1716542451772,20.87,0.15,1,0
1716542451863,20.87,0.15,1,0
1716542451954,20.87,0.15,1,0
1716542452061,20.87,0.15,1,0
1716542452163,20.86,0.17,1,0
1716542452263,20.86,0.15,1,0
1716542452364,20.87,0.15,1,0
1716542452459,20.87,0.15,1,0
1716542452560,20.87,0.15,1,0
1716542452653,20.87,0.15,1,0
1716542452752,20.87,0.15,1,0
1716542452855,20.87,0.15,1,0
1716542452961,20.86,0.15,1,0
1716542453054,20.86,0.15,1,0
1716542453164,20.86,0.15,1,0
1716542453264,20.86,0.15,1,0
1716542453365,20.86,0.15,1,0
1716542453460,20.86,0.15,1,0
1716542453565,20.86,0.15,1,0
1716542453654,20.86,0.15,1,0
1716542453753,20.86,0.15,1,0
1716542453865,20.86,0.15,1,0
1716542453963,20.86,0.15,1,0
1716542454058,20.86,0.15,1,0
1716542454163,20.86,0.15,1,0
1716542454263,20.86,0.15,1,0
1716542454382,20.86,0.15,1,0
1716542454461,20.86,0.15,1,0
1716542454561,20.86,0.15,1,0
1716542454661,20.86,0.15,1,0
1716542454762,20.86,0.15,1,1022
1716542454862,20.86,0.15,1,1022
1716542454973,20.86,0.15,1,1022
1716542455063,20.87,0.15,1,1022
1716542455164,20.86,0.15,1,1022
1716542455263,20.86,0.15,1,1022
1716542455365,20.86,0.15,1,1022
1716542455460,20.86,0.15,1,1022
1716542455559,20.86,0.15,1,1022
1716542455667,20.86,0.15,1,1022
1716542455763,20.86,0.15,1,1022
1716542455858,20.86,0.15,1,1022
1716542455963,20.85,0.15,1,0
1716542456053,20.85,0.15,1,0
1716542456171,20.85,0.15,1,0
1716542456263,20.85,0.15,1,1022
1716542456358,20.85,0.15,1,1022
1716542456468,20.85,0.15,1,1022
1716542456553,20.85,0.15,1,0
1716542456662,20.85,0.15,1,0
1716542456774,20.85,0.15,1,0
1716542456859,20.85,0.15,1,0
1716542456961,20.85,0.15,1,0
1716542457056,20.85,0.15,1,0
1716542457167,20.85,0.15,1,0
1716542457263,20.85,0.15,1,1022
1716542457358,20.85,0.15,1,1022
1716542457469,20.85,0.15,1,1022
1716542457555,20.85,0.15,1,0
1716542457656,20.85,0.15,1,0
1716542457759,20.85,0.15,1,0
1716542457882,20.85,0.15,1,0
1716542457964,20.86,0.15,1,0
1716542458059,20.86,0.15,1,0
1716542458163,20.86,0.15,1,0
1716542458263,20.86,0.15,1,0
1716542458364,20.86,0.15,1,0
1716542458459,20.86,0.15,1,0
1716542458557,20.86,0.15,1,0
1716542458665,20.86,0.15,1,0
1716542458760,20.86,0.15,1,1022
1716542458864,20.86,0.15,1,1022
1716542458958,20.86,0.15,1,1022
1716542459060,20.86,0.15,1,0
1716542459162,20.86,0.15,1,0
1716542459264,20.86,0.15,1,0
1716542459359,20.86,0.15,1,0
1716542459454,20.86,0.15,1,0
1716542459560,20.86,0.15,1,0
1716542459656,20.86,0.15,1,0
1716542459773,20.86,0.15,1,0
1716542459857,20.86,0.15,1,0
1716542459969,20.86,0.15,1,0
1716542460064,20.86,0.15,1,0
1716542460159,20.86,0.15,1,0
1716542460264,20.85,0.15,1,0
1716542460362,20.86,0.15,1,0
1716542460463,20.85,0.15,1,0
1716542460552,20.85,0.15,1,0
1716542460662,20.85,0.15,1,0
1716542460774,20.85,0.15,1,0
1716542460865,20.86,0.15,1,0
1716542460964,20.85,0.15,1,0
1716542461056,20.86,0.15,1,1022
1716542461164,20.86,0.15,1,1022

### To Reproduce

log MSP_ANALOG to a file

### Expected behavior

see only 0 values for RSSI

### Support ID

```txt
ac55bd4c-39f9-4efc-a896-09750b11d683
```

### Flight controller

Velox Cine F7

catkira, 2024-05-24:
it is also visible in the GUI
https://github.com/betaflight/betaflight/assets/13425558/9d54b19b-1acf-44d9-ada9-4e7822a83880

catkira, 2024-05-24:
I updated my ELRS RX module (Radiomaster RP1 2.4 GHz) to the latest 3.4.1 firmware, I did not change anything.

catkira, 2024-05-24:
I have also filed a ELRS bug report, because I think it could be ELRS related:
https://github.com/ExpressLRS/ExpressLRS/issues/2730

catkira, 2024-05-25:
I think it would be interesting to see these debug messages from crsf.c
```c
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_UPLINK, 0, stats.uplink_RSSI_1);
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_UPLINK, 1, stats.uplink_RSSI_2);
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_UPLINK, 2, stats.uplink_Link_quality);
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_UPLINK, 3, stats.rf_Mode);

    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_PWR, 0, stats.active_antenna);
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_PWR, 1, stats.uplink_SNR);
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_PWR, 2, stats.uplink_TX_Power);

    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_DOWN, 0, stats.downlink_RSSI);
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_DOWN, 1, stats.downlink_Link_quality);
    DEBUG_SET(DEBUG_CRSF_LINK_STATISTICS_DOWN, 2, stats.downlink_SNR);
```
can anybody tell me how I can enable them?

McGiverGim, 2024-06-13:
Simply go to the Blackbox tab and enable this debug mode.

haslinghuis, 2024-06-13:
Is this issue specific with ELRS 3.4.1 or did it happen on previous firmware?

catkira, 2024-06-13:
also with previous versions

closed 2026-09-26 by blckmn