betaflight/betaflight #13658: LED - Armed Mode

wjk1129, 2024-05-29:
### Describe the bug

We used to have 3 LEDs indicate Armed (Red) and Dis-Armed (Green) to say when the Drone was safe to pickup. 
Recent versions (including latest version DL'd 3 days ago)  the Armed LED function does not work. This has been going on for at least 6 months.  I do know about setting a LED in the modes selection screen. As a fall back, we are turning all the LEDs off when disarmed.  Thank you to anyone willing see what is the problem. 

### To Reproduce

Use programmable LED strip. We have 8 LEDs last attempt was 2 LED for Arm State, 2 for Red, 2 White, 2 for Blue.  Generally we like to use the battery state as a couple of LEDs.  The ArmState LEDs are not switching as we flip the arm switch on the controller. Arm state is shown correctly in the mode section of Beta Flight.

### Expected behavior

When the arm state switch on transmitter is switched, LEDs should switch from Green to Red.

### Configurator version

10.10.10

ASDosjani, 2024-05-29:
> @ASDosjani 🙃

I will check

CheeseAndChips, 2024-06-19:
I could not reproduce the issue, colors were correctly reacting to arm state. Tested on 7075ed6 commit and version 4.5.0

haslinghuis, 2024-06-19:
@wjk1129 without support ID provided as requested when opening the issue we don't have required information to resolve the issue.
Closing as we could not reproduce the issue.

closed 2024-06-19 by haslinghuis