betaflight/betaflight #13664: FURYF4OSD SoftSerial and SmartAudio don't work on 4.5

Hochmueller, 2024-05-31:
### Describe the bug

I used the LED pin of the FURYF4OSD to control  the TBS Unify Pro HV by remapping the pin to a softserial by:
4.4.3:
```
resource LED_STRIP none
resource SERIAL_TX 11 A00
```

4.5.0:
```
resource LED_STRIP none  
resource SOFTSERIAL_TX 1 A00 
```

Since I updated my FURYF4OSD from 4.4.3 to 4.5.0 I can't get the SmartAudio on a softserial to work anymore.  Following you can find the cli dump of 4.4.3 and 4.5.0:
4.4.3:
[clidump4.4.3.txt](https://github.com/user-attachments/files/15512712/clidump4.4.3.txt)
4.5.0:
[clidump4.5.0.txt](https://github.com/user-attachments/files/15512713/clidump4.5.0.txt)

On 4.4.3 the Video Transmitter page shows a valid connection:
[image]
When I check the serial and timer command I get:
```
# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 2 115200 115200 0 115200
serial 30 2048 115200 57600 0 115200

# timer
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer B09 AF3
# pin B09: TIM11 CH1 (AF3)
```

Compared to this on R4.5.0:
[image]

```
# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 2 115200 115200 0 115200
serial 30 2048 115200 19200 0 19200

# timer
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer B09 AF3
# pin B09: TIM11 CH1 (AF3)
```

I tried to update the serial 30 form 4.4.3 (serial 30 2048 115200 57600 0 115200) on 4.5.0(previously serial 30 2048 115200 19200 0 19200) as well but it did not fixed the problem. 

### To Reproduce

- Enable soft serial feature.
- unmap the LED_STRIP and set it to a soft serial by:
 ```
resource LED_STRIP none  
resource SOFTSERIAL_TX 1 A00 
```
- Setup Softserial 1 in the port tabs peripheral section to connect a smartaudio device (VTX TBS Smartaudio)
- Connect LED Pin of FURYF4OSD to a SmartAudo compatible device
- Go to the Video Transmitter view in Betaflight configurator an check SmartAudio status

### Expected behavior

- Valid connection on Smartaudio via softserial
- Be able to change vtx settings via fc

### Support ID

```txt
"35122aa9-5a30-44f9-9797-0eca733e1532"
```

### Flight controller

FURYF4OSD

### Other components

RX: iFlight ELRS 2.4G T on UART3
VTX: TBS Unify Pro HV
GPS: Geprc GEP M10 Series GPS Module on UART6

### How are the different components wired up (including port information)

ELRS RX on UART3
M10 GPS on UART6

haslinghuis, 2024-05-31:
Fixed in #13540

Hochmueller, 2024-05-31:
I can't agree. I tried to remote build the FW with the `OVERRIDE_SOFTSERIAL_BAUDRATE` flag on 4.5.0 (I checked it and the PR should be included in the release) but this did not fix the problem. Just to be sure I build it with the latest 4.5.0-maintanance as well but no success.

haslinghuis, 2024-05-31:
Hmmm. Smart Audio only requires 4800 baud so it should work without need for override 19200 restriction.

https://www.team-blacksheep.com/media/files/tbs_smartaudio_rev09.pdf

Hochmueller, 2024-05-31:
No luck. I still have the same issue.

freasy, 2024-06-01:
> No luck. I still have the same issue.

can you try this hex?
[betaflight_4.6.0_STM32F405_FURYF4OSD.zip](https://github.com/user-attachments/files/15520871/betaflight_4.6.0_STM32F405_FURYF4OSD.zip)

Hochmueller, 2024-06-01:
Yes sure. I tried it but still the same issue.

Lolig4, 2024-06-08:
I have the same problem

Konfigurator: 10.10.0 (c97deaf)
Firmware: 4.5.0 BTFL
Ziel: CLRA/CLRACINGF4(STM32F405)

`# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource SOFTSERIAL_TX 1 A00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SDI 1 A06
resource SPI_SDI 2 B14
resource SPI_SDI 3 C11
resource SPI_SDO 1 A07
resource SPI_SDO 2 B15
resource SPI_SDO 3 C12
resource CAMERA_CONTROL 1 B09
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
[BTFL_cli_backup_20240608_224942_CLRACINGF4.txt](https://github.com/user-attachments/files/15749656/BTFL_cli_backup_20240608_224942_CLRACINGF4.txt)
`

Lolig4, 2024-06-08:
I have the same problem

Konfigurator: 10.10.0 (c97deaf)
Firmware: 4.5.0 BTFL
Ziel: CLRA/CLRACINGF4(STM32F405)

```
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource SOFTSERIAL_TX 1 A00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SDI 1 A06
resource SPI_SDI 2 B14
resource SPI_SDI 3 C11
resource SPI_SDO 1 A07
resource SPI_SDO 2 B15
resource SPI_SDO 3 C12
resource CAMERA_CONTROL 1 B09
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
```

[BTFL_cli_backup_20240608_224942_CLRACINGF4.txt](https://github.com/user-attachments/files/15749667/BTFL_cli_backup_20240608_224942_CLRACINGF4.txt)

Hochmueller, 2024-06-18:
I did some tests. Basically I enabled USE_SMARTAUDIO_DPRINTF in vtx_smartaudio.c and routed it to a UART. (This uart is usually used for GPS).
Following the log:

```
smartAudioInit: OK
vtxSmartAudioInit 4 power levels recorded
smartAudioGetSettings
process: sending queue
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
autobaud: adjusting
autobaud: 4850
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
autobaud: adjusting
autobaud: 4900
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
autobaud: adjusting
autobaud: 4950
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
autobaud: adjusting
autobaud: now going down
autobaud: 4900
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
autobaud: adjusting
autobaud: 4850
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
autobaud: adjusting
autobaud: 4800
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
process: resending 0x1
```

I will try to log the smartaudio uart signals by a oscilloscope just to be sure that the baud rate works as expected.

Hochmueller, 2024-06-20:
Hello,
I did the measurements with the scope now. I figured out two problems:

1. It seems like the SW sets a pull up instead of a pull down. This builds a voltage divider between the pull down on the SmartAudio device in idle state:
4.4:
[image]
4.5:
[image]

1. The SW sends a 0 byte bevor it starts sending the frame. On 4.4 it seems like there is no valid start bit for this first byte:
[image]
while in 4.5 a valid start bit is generated by setting the signal high for a short amount of time:
[image]

By actually removing the start byte I could fix the problem, but I am not sure if this is a solution which would work on other serial devices as well. Bevor I will publish this solution I would like to understand why this start bit is generated now and was not generated bevor, maybe it is just a problem on the configuration.

haslinghuis, 2024-06-21:
See https://github.com/betaflight/betaflight/pull/13663#issuecomment-2142346035

Hochmueller, 2024-06-21:
I could identify the issue. In (https://github.com/betaflight/betaflight/pull/12929) the port options where adapted in vtx_smartaudio.c. Compared to previous version the option now does not include  SERIAL_BIDIR_NOPULL. This option is disabling the pull down/up in RX mode. 

In serial_softserial.c the pull option is not considered. It sets the pullup/down based on the SERIAL_INVERTED option. As Smartaudio is not inverted it will set a pullup. I assume the Smartaudio device has a pulldown included. So a voltage divider is build which explains the undefined voltage level on the Smartaudio line. 

Additional I figured out that the Smartaudio standard is violated at least on my TBS Unify Pro HV. The standard says:

"The SmartAudio line need to be low before a frame is sent. If the host MCU can’t handle this it can be done by
sending a 0x00 dummy byte in front of the actual frame."

This can be seen on the post above. As soon this is done the flight controller can't communicate to the Smartaudio device. But this can be fixed via the following cli command:

set vtx_softserial_alt = OFF 

I pushed a pull request to fix this issue. Test binaries are attached.

haslinghuis, 2024-06-21:
> This can be seen on the post above. As soon this is done the flight controller can't communicate to the Smartaudio device. But this can be fixed via the following cli command:

> set vtx_softserial_alt = OFF

Does it work with this setting ?  As OFF seems to be the default (in vtx.c:67)

https://github.com/betaflight/betaflight/blob/cff4c9cde354fdcb221756fd0d06af65e712ee92/src/main/io/vtx_smartaudio.c#L497-L510

Hochmueller, 2024-06-22:
> > This can be seen on the post above. As soon this is done the flight controller can't communicate to the Smartaudio device. But this can be fixed via the following cli command:
> 
> > set vtx_softserial_alt = OFF
> 
> Does it work with this setting ? As OFF seems to be the default (in io/vtx.c:67)
> 
> https://github.com/betaflight/betaflight/blob/cff4c9cde354fdcb221756fd0d06af65e712ee92/src/main/io/vtx_smartaudio.c#L497-L510

Yes in my tests it works. For some reasons I set vtx_softserial_alt to ON when updating my flight controller. So I was facing this two issues when debugging. But actually the main issue was the pullup while RX phase. 

https://github.com/betaflight/betaflight/blob/cff4c9cde354fdcb221756fd0d06af65e712ee92/src/main/drivers/serial_softserial.c#L126-L134

As I don't know if this is intentional in any other part of the code I decided to fix it by disabling any pull in RX phase as it was in R4.4

PR #13715 (2024-06-27): Fixes #13664 - Smartaudio don't work with Softserial 
Fixes #13664
In Smartaudio the softserial port option has changed from 4.4.x to 4.5.0. By this the Smartaudio line is in an unknown voltage state while no TX is running - this violates the Smartaudio standard. vtx_softserial_alt should be set to OFF when using Smartaudio. 

To test it I used the LED pin of the FURYF4OSD connected to TBS Unify Pro HV. Additional I tested it with serial port 6. Both tests where successful. More information can be found below. 

It is desirable to bring this fix into a patch release of 4.5.

If someone wants t [... cut]
commit 2ff71b0a (2024-06-27): Fixes #13664 - Smartaudio don't work with Softserial  (#13715)

Fixing issue #13664
Softserial port option has changed from 4.4.x to 4.5.0. By this the Smartaudio line is in an unknown voltage state while not TX is running - this violates the Smartaudio standard. vtx_softserial_alt should be set to OFF to use Smartaudio.
commit ec8c59b4 (2024-06-28): [4.5.1] Fixes #13664 - Smartaudio don't work with Softserial  (#13715) (#13734)
closed 2024-06-27 by haslinghuis