betaflight/betaflight #13665: Enabling camera control causes incorrect resource warning in Motors tab

morgalion, 2024-06-01:
### Describe the bug

I don't have permission to re-open this issue :
https://github.com/betaflight/betaflight/issues/11790
My comment on it was :

Hello,
I have the same problem with Betaflight 4.5.0, i want to use CAMERA_CONTROL feature for camera Caddx Rattel 1 on Holybro Kakute F7 (HBRO/KAKUTEF7).
I freed the motor 5 resource, cli related is:

resource motor 5 none
resource CAMERA_CONTROL 1 C09
set camera_control_key_delay = 210
set camera_control_internal_resistance = 270
save
I have the same warning as described before in motor tabs. Fix possible or just ignore the message ?
Thank you !

### To Reproduce

On HOLYBRO/KAKUTEF7 :
```
resource` motor 5 none
resource CAMERA_CONTROL 1 C09
set camera_control_key_delay = 210
set camera_control_internal_resistance = 270
save
```

Display an error message in motor tab

### Expected behavior

No message error in motor tab

### Support ID

```txt
ID : "05643408-9971-48f4-b8ac-08de64bf11b3"
```

### Flight controller

HOLYBRO KAKUTEF7

### Other components

Camera : Caddx Rattel 1

haslinghuis, 2024-06-01:
Try another timer then TIM8 and show resource show all

```bash
timer C09 AF2
save
```

morgalion, 2024-06-01:
Thank you, I changed timer with your CLI and the message is gone 🙂

haslinghuis, 2024-06-01:
Duplicate of #11790

closed 2024-06-01 by haslinghuis