betaflight/betaflight #13667: stm32f405 drone autopilot(ardupilot) function

koliamik, 2024-06-01:
### Is your feature request related to a problem? Please describe

[image]

### Describe the solution you'd like

Hi to all. I'm a software developer from Ukraine and I'm looking on GitHub, I'm asking for your help. I need to find a solution for the stm32f405 drone's autopilot function, where the parameters will remember the flight data and fly independently at the same speed. ; I haven't programmed in years and I can't figure out how to write the code. maybe the code is written, tell me how to find it and put this function under AUX. Thank you for your attention, helping me is helping Ukraine. As I understand it, on the principle of an airplane, the autopilot is inputting flight data and converting it into a constant and reading gyroscope data, and GPS data, but I would like to solve this problem without it. well, there is also an idea that this function should work regardless of the loss of the signal with the remote control.

### Describe alternatives you've considered

ArduCopter this program is an example of how to write and need to register the functionality of fpv.

### Other information

[ArduCopter](https://github.com/ArduPilot/ardupilot/tree/51c77fe098dc546654784c075010a8a74533e830/ArduCopter) 
```

#include "Sub.h"

// checks if we should update ahrs/RTL home position from the EKF
void Sub::update_home_from_EKF()
{
// exit immediately if home already set
if (ahrs.home_is_set()) {
return;
}

// special logic if home is set in-flight
if (motors.armed()) {
set_home_to_current_location_inflight();
} else {
// move home to current ekf location (this will set home_state to HOME_SET)
if (!set_home_to_current_location(false)) {
// ignore this failure
}
}
}

// set_home_to_current_location_inflight - set home to current GPS location (horizontally) and EKF origin vertically
void Sub::set_home_to_current_location_inflight()
{
// get current location from EKF
Location temp_loc;
Location ekf_origin;
if (ahrs.get_location(temp_loc) && ahrs.get_origin(ekf_origin)) {
temp_loc.alt = ekf_origin.alt;
if (!set_home(temp_loc, false)) {
// ignore this failure
}
}
}

// set_home_to_current_location - set home to current GPS location
bool Sub::set_home_to_current_location(bool lock)
{
// get current location from EKF
Location temp_loc;
if (ahrs.get_location(temp_loc)) {

// Make home always at the water's surface.
// This allows disarming and arming again at depth.
// This also ensures that mission items with relative altitude frame, are always
// relative to the water's surface, whether in a high elevation lake, or at sea level.
temp_loc.alt -= barometer.get_altitude() * 100.0f;
return set_home(temp_loc, lock);
}
return false;
}

// set_home - sets ahrs home (used for RTL) to specified location
//  returns true if home location set successfully
bool Sub::set_home(const Location& loc, bool lock)
{
// check if EKF origin has been set
Location ekf_origin;
if (!ahrs.get_origin(ekf_origin)) {
return false;
}

// set ahrs home (used for RTL)
if (!ahrs.set_home(loc)) {
return false;
}

// lock home position
if (lock) {
ahrs.lock_home();
}

// return success
return true;
}

```

SteveCEvans, 2024-06-01:
So, you’re looking for jam proof (both RC link and GPS) inertial guidance?

koliamik, 2024-06-01:
just so that it flew straight and the failsafe did not work. and there is no GPS dacha. only one has an accelerometer, that is, if the wind is so that it does not change the position of the drone. 
tut niby i ne potribnyy avtopilot tut prosto shchob vin ne vymykav dvyhuny i letiv po zadaniy traeyiktoriyi
here it seems that the autopilot is not needed here, just so that it does not turn off the engines and flies along the given trajectory

Quick-Flash, 2024-06-01:
This is not possible without GPS. Yes accelerometer can tell you if you started moving, but the ACC will always have bias and never be quite accurate, those small inaccuracies quickly add up and it becomes near impossible to know your location after a second or two without GPS. So it would I suppose be possible to integrate the ACC data and then leak it pretty darn quick so as to not let things get out of control, but that could at best only slightly help fight the wind automatically for you. TLDR: without GPS this is not possible.

ledvinap, 2024-06-01:
https://habr.com/ru/articles/346662/, https://en.wikipedia.org/wiki/Habr

I'm afraid it's too easy to help wrong side

closed 2024-06-03 by haslinghuis