betaflight/betaflight #13673: bf4.5固件，在betafpv 5A aio上无法修改电机的方向

gaodianzhuo, 2024-06-06:
### Describe the bug

bf4.5在betafpv 5A aio上无法修改电机的方向，这是我的实测，根据飞友报告说高于4.4.0都不行

祝好！

### To Reproduce

刷新固件

### Expected behavior

能够改变电机的旋转方向

### Support ID

```txt
不懂
```

### Flight controller

betafpv 5A aio

yjtang, 2024-06-10:
Translation: with BF4.5 firmware, Betafpv 5A AIO cannot change motor rotation direction.

haslinghuis, 2024-06-10:
Please try 4.5.1-dev

gaodianzhuo, 2024-06-13:
> Please try 4.5.1-dev

thanks,i flash BF4.4.0 and everything works fine now.

closed 2024-06-13 by haslinghuis