betaflight/betaflight #13689: HIFICONRCF722 updated BF4.5.0 can't find dataflash

AtherisFPV, 2024-06-15:
### Describe the bug

HIFICONRCF722 updated from BF4.4.3 to BF4.5.0, then the latest version can't find dataflash. But the BF4.4.3 version everything is work fine.
[image]

### To Reproduce

DFU Mode, used the BF configurator 10.10.0 to update the flight controller.tried many times included the "core only"mode, but dataflash still error.even i tried flash the EMUflight and then reflash Betaflight , but still. 

### Expected behavior

HIFICON F722 in BF4.5.0 version can work properly. Thank you, handsome Bro😘 

### Support ID

```txt
Id: "e538617a-60fb-499d-8e3c-fd3c2cb3814c
```

### Flight controller

HFOR/HIFIONRCF7(STM32F7X2)

haslinghuis, 2024-06-15:
Try adding `FLASH FLASH_W25Q128FV` in custom defines while flashing

AtherisFPV, 2024-06-18:
Thanks Bro! I flow your instructions did it, the databox is back to normal. Please refer to the screenshots for my operation. Thank you~❤🚀
[image]
[image]