betaflight/betaflight #13700: Null Deref in CLI gpspassthrough

HorizontalCar, 2024-06-18:
### Describe the bug

If `featureIsEnable(FEATURE_GPS)`  is false, `gpsInit` is not called and therefore , `gpsPort` is never initialized. However in [gpspassthrough](https://github.com/betaflight/betaflight/blob/660018b1de0c32ee60d8321f301d36fb1fd097f9/src/main/cli/cli.c#L3599) CLI, it never checks that this is the case and attempts to dereference `gpsPort` resulting in a null pointer dereference.. https://github.com/betaflight/betaflight/blob/660018b1de0c32ee60d8321f301d36fb1fd097f9/src/main/io/gps.c#L2494

### To Reproduce

**In CLI mode:**

```
Entering CLI Mode, type 'exit' to return, or 'help'

# gpspassthrough
```

### Expected behavior

Correctly NULL check `gpsPort`.

### Support ID

```txt
N/A
```

### Flight controller

N/A

### Add any other context about the problem that you think might be relevant here

I've been testing automated firmware testing tools on Betaflight firmware, so I am not using the firmware directly. Instead, I am emulating it in a testing environment and manually triaging the crashes I find. During my testing, I found inputs that trigger a null pointer dereference in CLI mode.

ctzsnooze, 2024-09-05:
@ledvinap how would we fix this?

ledvinap, 2024-09-05:
simply abort gpsEnablePassthrough if port of null (or not open)

closed 2024-09-06 by haslinghuis