betaflight/betaflight #13701: Null dereference in cliAux

HorizontalCar, 2024-06-18:
### Describe the bug

`nextArg(const char *currentArg)` assumes `currentArg` is non-`NULL`, however, the `cliAux` seems to assume that `nextArg` will return `NULL` if passed a `NULL` input as shown. 

https://github.com/betaflight/betaflight/blob/660018b1de0c32ee60d8321f301d36fb1fd097f9/src/main/cli/cli.c#L1090C10-L1093C25

As a result, if `currentArg` is `NULL` a null pointer dereference occurs at:

https://github.com/betaflight/betaflight/blob/660018b1de0c32ee60d8321f301d36fb1fd097f9/src/main/cli/cli.c#L998

### To Reproduce

**In CLI mode:**

```
Entering CLI Mode, type 'exit' to return, or 'help'

# aux aI
```

### Expected behavior

Do not crash when using CLI aux incorrectly

### Support ID

```txt
N/A
```

### Flight controller

N/A

### Add any other context about the problem that you think might be relevant here

I've been testing automated firmware testing tools on Betaflight firmware, so I am not using the firmware directly. Instead, I am emulating it in a testing environment and manually triaging the crashes I find. During my testing, I found inputs that trigger a null pointer dereference in CLI mode.

haslinghuis, 2024-06-19:
Result:

```bash
# aux aI
aux 0 0 0 900 900 0 0
```

ledvinap, 2024-06-20:
@haslinghuis : null pointer dereference is undefined behavior. this problem should not be ignored

nerdCopter, 2024-07-16:
i do not know how to reproduce, even copy/pasting or manually "typing" null
https://unicode-explorer.com/c/0000
[image]

ledvinap, 2024-07-17:
The lines in @HorizontalCar do not match cliAux, first problem is in https://github.com/betaflight/betaflight/blob/master/src/main/cli/cli.c#L1206-L1215

The code will work on ARM (flash is mapped to 0x0 / NULL, so it will simply read vector table until some space or '\0' byte is found. First byte of flash contains top of stack (in little endian), `00 00 02 20` on F7, so `nextarg` will fail quite gracefully.
Changing initial stack address can cause very interesting consequences ..

ledvinap, 2024-07-17:
(all targets with estack defined  as `- 8` will probably fail / report different error)
```
./src/link/stm32_h750_common.ld\06:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM);    /* end of RAM */
./src/link/stm32_ram_h743.ld\067:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - 8; /* Reserve 2 x 4bytes for info across reset */
./src/link/stm32_flash_g4_split.ld\016:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - _Hot_Reboot_Flags_Size;    /* end of RAM */
./src/link/stm32_flash_h743_2m.ld\060:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - 8; /* Reserve 2 x 4bytes for info across reset */
./src/link/stm32_flash_f4_split.ld\016:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - _Hot_Reboot_Flags_Size;    /* end of RAM */
./src/link/stm32_h723_common.ld\06:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM);    /* end of RAM */
./src/link/stm32_flash_h563_2m.ld\048:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - 8; /* Reserve 2 x 4bytes for info across reset */
./src/link/stm32_flash_f7_split.ld\015:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM);    /* end of RAM */
./src/link/stm32_flash.ld\015:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM);    /* end of RAM */
./src/link/at32_flash_f4_split.ld\016:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - _Hot_Reboot_Flags_Size;    /* end of RAM */
./src/link/stm32_flash_h7a3_2m.ld\057:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - 8; /* Reserve 2 x 4bytes for info across reset */
./src/link/stm32_flash_h723_1m.ld\058:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM) - 8; /* Reserve 2 x 4bytes for info across reset */
./src/link/stm32_h730_common.ld\06:_estack = ORIGIN(STACKRAM) + LENGTH(STACKRAM);    /* end of RAM */