betaflight/betaflight #13717: Support for WLED

salembegic, 2024-06-22:
### Is your feature request related to a problem? Please describe

I just recently got into FPV and I wanted to customize my first quad. I soon noticed that LED support in Betaflight is quite basic, with the maximum of 32 LEDs and with most options being simple static colors or blinking...

### Describe the solution you'd like

It would be great if we could add more than 32 LEDs, choose custom presets, color combinations, effects, and so on, but I also know that LEDs aren't really a priority and that implementing those features would mean changing a lot legacy code, and it would also increase processor usage, but I think I found a solution that would work to give drones more unicorn barf while keeping the old sistem for basic users.

### Describe alternatives you've considered

Even before Betaflight I used an open source software called [Wled](https://github.com/Aircoookie/WLED) on a ESP32 board (2-3€ on AliExpress) to control the led strips around my house. It can control thousands of LEDs at once and it has more features than I would ever need, so I did some testing and it seems to work perfectly running off 5V BEC. It already works great if you just want to set it and forget it, but I would like it to be more integrated in the whole system so that I can change the settings using my controller. I did some research and found out that it has a serial API so it should be possible to configure it just like any other UART device. In my opinion it would be easiest if Betaflight just offered the possibility to change presets using the API, and the configuration of the LEDs can be left to WLED interface. I am not super good with programming, so I'm not sure how hard this would be, but if you have any questions I will try to help in any way I can.

SteveCEvans, 2024-06-22:
We have an option for 64 LEDs. The issue is really the time taken to calculate the colour values, and to a lesser extent the storage needed for the buffer which amounts to nine bytes per LED.

salembegic, 2024-06-22:
I absolutely understand, and the current version is great at what it does but I want to use some COB LED strips I have lying around, and those are like 100-200 LEDs per meter. The only way would be to connect them in parallel, but that probably wouldn't look the best. And honestly, animations in WLED are amazing. I'm not the best at setting them up, but there are some guys on Youtube like Chris Maher that do an amazing job

closed 2024-07-30 by github-actions[bot]