betaflight/betaflight #13720: Mamba H743 v2 boots to DFU mode when calibrating accelerometer

NspaceCITY713, 2024-06-24:
### Describe the bug

Any time I calibrate the accelerometer, flight controller reboots into DFU mode. It does calibrate and save the entry but kicks it into DFU mode, requiring me to restart. Used to happen maybe 1 out 5 tries, now on 4.6.0 it does it every single time. 

### To Reproduce

calibrate accelerometer and boots back into dfu

### Expected behavior

calibrate, save and reconnect 

### Support ID

```txt
6c37c7d0-80a1-4fb0-be3c-7a1290e523ae
```

### Flight controller

Diatone Mamba H743 v2

### Other components

Mamba f65 MK4 ESC, o3 Air unit, Happymodel ep1 rx 

### Add any other context about the problem that you think might be relevant here

Back on 4.4 it would do this as well, but it wouldn't happen all the time. Maybe 1 out of 5 tries when calibrating accelerometer. But it would also randomly happen when trying to save changed settings, instead of saving and rebooting, every now and then it would go into DFU. Now it is only occurring when calibrating accelerometer and every time.

NspaceCITY713, 2024-06-24:
[image]

haslinghuis, 2024-06-24:
Firmware lockup. Can you try detaching other peripherals to further isolate the issue. Suggest starting with receiver.

NspaceCITY713, 2024-06-25:
> Firmware lockup. Can you try detaching other peripherals to further isolate the issue. Suggest starting with receiver. 

Yes I can, give me a few. I've got happymodel ep1 rx on quad, and a happymodel openlager as I was having blackbox issue with flash drive (windbond w25n01gv). Quad flies fine, just gets annoying trying to setup and it kicks it to dfu once in while.

NspaceCITY713, 2024-06-25:
So I have a second flight controller, same exact one brand new in box,and I've come to find the issue isn't related to calibrating the accelerometer, but more so the save and reboot. It happens randomly when changing a setting. Tried the new flight controller with the firmware it came with, 4.3, same issue. Board has zero peripherals to it. Tried 4.6.0, same issue. Did some digging in old posts and found a similar issue with diatone h7 board, #11516 , https://github.com/betaflight/betaflight/issues/11516 ,  where they had issues with save and reboot, originating from osd_displayport=msp. I have the same issue but changing osd_DisplayPort=msp to =auto or =off doesn't solve the issue like some stated in that post. Originally thought it was accelerometer calibration but it's due to the save/reboot after calibrating bc it happens sometimes also when saving a setting.

kedeng, 2024-07-15:
I found this issue related to MSP messages. When the flight controller is connected via USB, device still use the MSP port to send MSP messages. While the user is trying to save something, the MSP message gets truncated or corrupted. If there is a character equal to 'R' ("get reboot_character" through cli ), the FC will go into DFU mode.

kedeng, 2024-07-16:
> I found this issue related to MSP messages. When the flight controller is connected via USB, device still use the MSP port to send MSP messages. While the user is trying to save something, the MSP [...]

I think it's very dangerous to trigger the FC into DFU mode by sending just a single character. Especial in ARM mode, we can't ensure MSP device send right MSP messages, the serial port may disturb and send unexpected reboot character.

NspaceCITY713, 2024-07-26:
Any suggestions on a fix ?

kedeng, 2025-04-08:
@haslinghuis , I believe this bug has been fixed by by https://github.com/betaflight/betaflight/pull/14001

haslinghuis, 2025-04-08:
Fixed in #14001

closed 2025-04-08 by haslinghuis