betaflight/betaflight #13721: GPS Altitude zero'ing out on arming

dominykasz, 2024-06-24:
### Describe the bug

Hi, according to some people when you have your altitude source set to GPS_ONLY, it should display your above sea level gps altitude and not above the take off point.

Well for me once I start the drone while it's dissarmed, it does show GPS altitude above sea level (I have checked, it is correct altitude), but once I arm my drone it resets it to 0m so basically from take off point, watch the video to see.

https://github.com/betaflight/betaflight/assets/41077740/95a2d04c-bd2f-4830-b1da-3028ab433071

### To Reproduce

1. Go into Betaflight CLI
2. Set altitude source to GPS Only by doing "set altitude_source = GPS_ONLY"
3. Type "save"
4. Double check that source is GPS Only by "get altitude_source" which should result in "altitude_source = GPS_ONLY"
5. Start your drone by plugging in the battery
6. Turn on VTX, don't arm the drone
7. Check your altitude, should show your real accurate above sea level alittude so for me it's 108m
8. Arm your drone
9. Check altitude again, it should show 0m now
10. While flying the altitude will always be shown in respect to take off point.

### Expected behavior

5. Start your drone by plugging in the battery
6. Turn on VTX, don't arm the drone
7. Check your altitude, should show your real accurate above sea level alittude so for me it's 108m
8. Arm your drone
9. Check altitude again, it should show 0m now
10. While flying the altitude will always be shown in respect to take off point.

### Support ID

```txt
a80159ec-fd60-4a5d-a9ea-13c7785a1d31
```

### Flight controller

Mamba MK4 H743 V2 Stack

### Other components

GPS In question is  Mateksys GNSS & Compass M10Q-5883

https://www.mateksys.com/?portfolio=m10q-5883

### How are the different components wired up (including port information)

GPS Is wired up to UART 8.

Wiring goes like this: 
    M10Q-5883  5V to Flight controller 4~9V
    M10Q-5883  RX to Flight controller UART_TX
    M10Q-5883  TX to Flight controller UART_RX
    M10Q-5883  CL to Flight controller SCL
    M10Q-5883  DA to Flight controller SDA
    M10Q-5883  G  to Flight controller GND

haslinghuis, 2024-06-24:
Duplicate of #13360

dominykasz, 2024-06-24:
> Duplicate of #13360

How is this a duplicate? Could you at least link to where in that other issue it relates to my issue? @haslinghuis

haslinghuis, 2024-06-24:
After talk on discord discovered issue is with GPS data zeroing out once armed and returns on disarm.
Baro is not working due to coating applied.

ctzsnooze, 2024-09-04:
Yes, current behaviour is that after arming, the display should be relative altitude compared to takeoff point, not absolute altitude.  Absolute altitude is shown only while disarmed.

closed 2026-09-26 by blckmn