betaflight/betaflight #13728: Stop accepting new architectures until existing code is consolidated

ledvinap, 2024-06-26:
### Is your feature request related to a problem? Please describe

Handling of new architectures is getting out of control. 
https://github.com/betaflight/betaflight/pull/13709# is nice example - it is following BF conding stanrards, but reveals tons of code duplication in existing code base. If this is allowed to continue, it will cause huge problems in future

### Describe the solution you'd like

Declare moratorium on new architectures. Accept new targets only after existing code is cleaned up.
Vendors with new targets can sponsor cleanup/help with it.

### Describe alternatives you've considered

1. Continue as usual. Fix problems as they surface
2. Start migrating to RTOS immediately. RTOS will alleviate some problems (using HAL from RTOS, simplifying some low-level code)

blckmn, 2024-06-26:
@ledvinap agreed. However whilst we may implement a moratorium - we will actually need a way forward. 

My goal is that /src/main/* has no reference to MCU or other hardware at all. There should not be any "#if defined(STMXXX)" in this folder.

Therefore there should be a structure for moving forward, and it will assist in the preparation for implementing RTOS.

Perhaps (for discussion):

/lib <- all library and support files. Should be minimal edits from developers, kept as close as possible to manufacturers, so that updates are easier.
/src/main <- main betaflight code. No hardware references. As portable as possible.
/src/device <- hardware code, including linking scripts, startup files and the like and any MCU specific driver requirements, flash drivers, baro, gyro and other hardware implementations. Would include the target directory.

e.g. 
/src/device/drivers/baro
/src/device/drivers/flash
/src/device/drivers/acc_gyro
/src/device/mcu/stm32/startup
/src/device/mcu/stm32/link
/src/device/target

Whilst the explosion is largely due to STM clones at the minute, there is very much a non STM clone we are working on that has multiple cores and some major differences! Tidying up the clones is definitely doable, but it is arguably just as bad to have a litany of #ifdef #elif #endif for the sake of combining code into one file. There is a compromise of course, and we can move to having a better interface and implementation layout so duplication is removed, but files provide the separation between MCUs where specific MCU implementation code is required.

ledvinap, 2024-06-28:
> My goal is that /src/main/* has no reference to MCU or other hardware at all. There should not be any "#if defined(STMXXX)" in this folder.

It won't be easy. Doing it cleanly would require some common HAL abstraction layer. But too generic abstraction will cause performance bottlenecks and writing BF-specific HAL is hard (simple IO abstraction took considerable effort and time. But it maybe be a over-optimised). So some compromise will be necessary.

Some code can be improved by using 'concept' instead of 'mcu family' for #ifdefs. For example stream vs channel based DMA, AF register usage, ...

Another problem is that logic and hardware dependent layer is mixed in some code. `camera_control.c` is one example, with logic duplicated in multiple places. Ideally, hardware-dependent functions would be extracted into separate file (but that file may be common for most targets).

> Tidying up the clones is definitely doable, but it is arguably just as bad to have a litany of #ifdef #elif #endif for the sake of combining code into one file. 

It is ugly, but it has very important advantage - all code versions are in one place. Multiple bugs were/are caused by UART logic residing in multiple files. Devs usually check only target they use and it is almost impossible to check it on review (unless you have intimate knowledge of driver implementation).
I'd prefer keeping logic in single file, either with concept #defines in target-dependent file, or simple stubs, possibly in separated into *-mcu.c file, but sharing this file between similar/all architectures where possible.

> /lib <- all library and support files. Should be minimal edits from developers, kept as close as possible to manufacturers, so that updates are easier.
> /src/main <- main betaflight code. No [...]

There is some code that is quite tightly coupled to MCU (SPI/QSPI flash, probably DSHOT .. ). We can progress incrementally and see what happens.

> Whilst the explosion is largely due to STM clones at the minute, there is very much a non STM clone we are working on that has multiple cores and some major differences!

For now, it may be better to simple run BF in single RTOS thread on multicore MCU (https://github.com/BrainFPV/betaflight)

Maybe use could some automated tool for duplicated code detection and start local refactoring from it ...

blckmn, 2024-07-07:
> It won't be easy. Doing it cleanly would require some common HAL abstraction layer. But too generic abstraction will cause performance bottlenecks and writing BF-specific HAL is hard (simple IO [...]

The risk is that it is highly optimised and we will need to back away from that in order to progress. but we need not do it immediately, just to keep it in mind.

> Some code can be improved by using 'concept' instead of 'mcu family' for #ifdefs. For example stream vs channel based DMA, AF register usage, ...
 
This should definitely be applied and most certainly for similar architectures. I feel strongly that if we ensure we do not have MCU or device specific logic mixed in with BF logic this will be a good step.

> Another problem is that logic and hardware dependent layer is mixed in some code. `camera_control.c` is one example, with logic duplicated in multiple places. Ideally, hardware-dependent functions would be extracted into separate file (but that file may be common for most targets).

This is where it is important to separate device and BF logic. We will need to progress slowly, and carefully - but we need that ultimate goal. 

> It is ugly, but it has very important advantage - all code versions are in one place. Multiple bugs were/are caused by UART logic residing in multiple files. Devs usually check only target they use [...]

Whether in the same file, or separate files, if the developer does not check all `define` logic paths, the same issue persists. Admittedly it will be easier to spot as a reviewer in the one file... but the problem is still the developer not checking all compilation paths. File size growth can also impact this ability.

> There is some code that is quite tightly coupled to MCU (SPI/QSPI flash, probably DSHOT .. ). We can progress incrementally and see what happens.
 
I like the idea of small incremental changes. Progression towards the goal.

Key for me is for /src/main to be free of device logic and portable as possible. Even to the extent we do not reference types from the various (and growing in number) MCU libraries.

LYNHQQ, 2025-07-18:
We could consider using the svd2rust tool to generate the code