betaflight/betaflight #13741: front " som·er·sault" movement causes quad / gyro to freak out

DroneShotFPV, 2024-07-01:
### Describe the bug

Before stepping away from the hobby a bit, I had this issue, assumed it was my PYRODRONEF4 board that was the issue, mounted upside down and 180, so didn't pay much mind to it... was running BF 4.1 at the latest (was happening even prior to 4.1)... I got back into the hobby 4 weeks ago, new(ish) build of brand new parts that had a while, Brain FPV RADIX (not v2) and TMOTOR F35 individual ESC's... BF 4.5, same issue. 
Full stick deflection in the upper right corner result in a the quad sounding and acting as it's spazzing out not knowing what is happening and sounds like a panic, then recovery from the flip. I get more air loss than would on a standard front flip and sometimes when it comes out of this simple move it's flying quickly in the direction it "recovered" from. 
Standard straight up full deflection front flips are fine and as expected / normal. Back flips, rolls, etc, ALL 100% fine, but FULL deflection to the upper right corner (haven't tested opposite corner) result in the freak out every single time, it's not sporadic, and I guarantee it can be reproduced each time. I am building another BF quad as I sold several before stepping away for a couple years to ensure it's standard across all iterations and is in the software control not the hardware. 
As previously stated, first time I noticed was a PYRODRONEF4 board which uses a different GYRO than the BrainFPV RADIX.
All my KISS setups do not do this and perform this maneuver without issue, including my old KISS V2 board, and new Fettec F7 board. 
The BF setup in question currently (building 2 more to also add to report just for data sake) is:
Rotor Riot 2306 2450kv motors
TMOTOR F35 ESC's 
BrainFPV RADIX FC
TBS Crossfire NanoRX
TBS Unify Pro V3 5v
Brain FPV PDB
Rates. BF rates: RC Rate 1.46 - Rate 0.75, EXPO 0.21.
Degrees rotation 1168
Tried Actual rates as well to see if rates affected it in any way, they did not
PIDS are set to default, as well as hand tuned. Sliders for filtering adjusted every which way that didn't burn my quad out.. lol
This one is a 4s build, previous with PYRODRONEF4 board was a 6s.
I have tried BF 4.5, 4.4, 4.3, 4.2 and 4.1
Any further requirements let me know and I will provide what I can.

### To Reproduce

Fly forward, throttle up, do a front flip, but do full stick deflection in the top right corner of the radio, not just a full forward deflection like a normal flip, but forward and upper right all at once.
Not sure if speed matters, but my rotation degree per second is 1168 with RC Rate 1.46, Rate 0.75 and EXPO 0.21

### Expected behavior

Gyro freaks out, quad sounds like the motors go into a frenzy and "recovers" in strange ways like it came out of a tail spin and trying to avoid crashing. If you are low enough to the ground when doing this, it will crash and hit the ground.

### Support ID

```txt
10343384-80f9-4169-bd43-bcf43a006c51
```

### Flight controller

BRAIN FPV RADIX

### Other components

TBS Unify Pro V3 5v
TBS Crossfire Nano RX
TMOTOR F35 ESC's (individual)
BrainFPV PDB

haslinghuis, 2024-07-01:
BrainFPV has it's own repo. Please create a support issue here: https://github.com/BrainFPV/betaflight

closed 2024-07-01 by haslinghuis

DroneShotFPV, 2024-07-01:
> BrainFPV has it's own repo. Please create a support issue here: https://github.com/BrainFPV/betaflight

Yes, but this also happens in other boards using Betaflight, as Indicated stating the PYRODRONEF4 and has happened since at least 4.1 that I can note.

haslinghuis, 2024-07-01:
Checkout [discord](https://discord.com/invite/gV4XSq3fus) for support.