betaflight/betaflight #13742: Full stick deflection in upper right corner causes GYRO to freak out.

DroneShotFPV, 2024-07-01:
### Describe the bug

Issue I am having on a PYRODRONEF4 board, OMNIBUSF7, and RADIX is as follows:
Full stick deflection in the upper right corner result in a the quad sounding and acting as it's spazzing out not knowing what is happening and sounds like a panic, then recovery from the flip. I get more air loss than would on a standard front flip and sometimes when it comes out of this simple move it's flying quickly in the direction it "recovered" from.
Standard straight up full deflection front flips are fine and as expected / normal. Back flips, rolls, etc, ALL 100% fine, but FULL deflection to the upper right corner (haven't tested opposite corner) result in the freak out every single time, it's not sporadic, and I guarantee it can be reproduced each time. I am building another BF quad as I sold several before stepping away for a couple years to ensure it's standard across all iterations and is in the software control not the hardware.
This is replicated on other Betaflight boards as I just tested another this morning with my OMNIBUSf7 build, as well as a PYRODRONEF4. and RADIX board.

Betaflight versions 4.1 - 4.5 ALL do the same thing.

### To Reproduce

throttle up, full stick deflection in upper right corner (nor straight front flip, just upper right corner of pitch stick)

### Expected behavior

Gyro freaks out, quad sounds like the motors go into a frenzy and "recovers" in strange ways like it came out of a tail spin and trying to avoid crashing. If you are low enough to the ground when doing this, it will crash and hit the ground.

### Support ID

```txt
05c55ab0-1020-4930-a47e-2215137751e4
```

### Flight controller

OMNIBUSF7

### Other components

TMOTOR VELOX 4in1 ESC
TBS Unify Pro V3
Crossfire Micro TX

haslinghuis, 2024-07-01:
- Duplicate of #13741
- Provide blackbox log
- Get support on Discord.

DroneShotFPV, 2024-07-01:
ESC firmware is 32.7 BLHeli32, No current BB Log. You say support on Discord, you're telling me this issue that exists across multiple hardware types and boards isn't a potential issue with firmware?

haslinghuis, 2024-07-01:
Please provide a blackbox log. Also try using DSHOT300.

closed 2024-09-10 by github-actions[bot]