betaflight/betaflight #13747: MAG enabling causes YAW angle reset

andreikovari, 2024-07-03:
### Describe the bug

On a custom board based on SPEEDYBEEF4 V4 (same IMU, same BARO), with an extra MAG (LIS2MDL).
Firmware loaded through configurator, 4.5.0, with the magnetometer tag added (original FC does not have a MAG installed).
With the default firmware (no MAG enabled) all works fine.
With the MAG enabled firmware, the yaw is being reset to 330~340 degrees, irrespective of the other angles.

1. Other functionalities look normal.

### To Reproduce

HW: SPEEDYBEEF4 V4 + MAG
Upload 4.5.0 with the magnetometer tag added.
Rotate to change the yaw angle.

### Expected behavior

Yaw angle follows the FC movement, does not sporadically reset.

### Support ID

```txt
51c3cc81-1451-4f97-8ece-14c8a792f201
```

### Flight controller

SpeedyBeeF4 V4 with added Magnetometer LIS2MDL

### Other components

Nothing else connected. 

### How are the different components wired up (including port information)

Magnetometer added on I2C bus.

YarosMallorca, 2024-07-06:
I can repro this with a GepRC GPS+Mag. Same issue...
I disabled Mag for now as Betaflight doesn't use it for much anyway.

ctzsnooze, 2024-09-04:
Was the mag oriented and. calibrated correctly, as per the wiki?
If not, it will return weird data.

YarosMallorca, 2024-09-05:
I tried calibrating the Mag as per wiki, even connecting to betaflight over ELRS so I could go further away from any potential interference (computers/metal...) but it didn't work.
I no longer have GPS  on my quad so I can't say if this has been fixed or not.

PeterJFrancoIII, 2025-03-20:
I can confirm, this is an ongoing issue.
Speedybee F405 V4 using both Beitian BN-880 w/HMC5883 as well as BZ-251 w/QMC5883. 

I’ve tried at least 30 attempts to Calibrate, all of which have failed.

Set Custom Orientation as per Wiki and the Yaw angle snaps back every time.

haslinghuis, 2025-03-20:
Please check 
- https://github.com/betaflight/betaflight/pull/13487
- https://betaflight.com/docs/wiki/release/Betaflight-4-5-Release-Notes#4-magnetometer-update

ORB1999, 2025-05-27:
> I can confirm, this is an ongoing issue. Speedybee F405 V4 using both Beitian BN-880 w/HMC5883 as well as BZ-251 w/QMC5883.
> 
> I’ve tried at least 30 attempts to Calibrate, all of which have failed.
> 
> Set Custom Orientation as per Wiki and the Yaw angle snaps back every time.

Were you finally able to solve the issue. I have a similar setup and have tried calibrating multiple times with no luck. I am also getting a YAW angle reset all the time.

PeterJFrancoIII, 2025-05-27:
> ORB1999

Never figured it out and my eyes were bleeding by the time I gave up.
I’m not a person that gives in easily either. 
This thing sucks.

ORB1999, 2025-05-27:
> ### Describe the bug
> On a custom board based on SPEEDYBEEF4 V4 (same IMU, same BARO), with an extra MAG (LIS2MDL). Firmware loaded through configurator, 4.5.0, with the magnetometer tag added (original FC does not have a MAG installed). With the default firmware (no MAG enabled) all works fine. With the MAG enabled firmware, the yaw is being reset to 330~340 degrees, irrespective of the other angles.
> 
> 1. Other functionalities look normal.
> 
> ### To Reproduce
> HW: SPEEDYBEEF4 V4 + MAG Upload 4.5.0 with the magnetometer tag added. Rotate to change the yaw angle.
> 
> ### Expected behavior
> Yaw angle follows the FC movement, does not sporadically reset.
> 
> ### Support ID
> 51c3cc81-1451-4f97-8ece-14c8a792f201
> ### Flight controller
> SpeedyBeeF4 V4 with added Magnetometer LIS2MDL
> 
> ### Other components
> Nothing else connected.
> 
> ### How are the different components wired up (including port information)
> Magnetometer added on I2C bus.
> 
> ### Add any other context about the problem that you think might be relevant here
> _No response_

Did you have any luck?

blckmn, 2026-09-26:
We're closing this as part of a clean-up of issues more than two years old. Sorry it sat here so long without a resolution, that's on us. Betaflight has changed a lot since this was opened, so if it's still relevant on the latest release, please open a new issue with current details and link back here.

closed 2026-09-26 by blckmn

PeterJFrancoIII, 2026-09-27:
Completely gave up. There was no solution despite months of attempts. Gpt, gemini and claude couldn’t helpPeter J Franco III