betaflight/betaflight #13751: Есть возможность из полетного контролера сделать наземный?

Win3273, 2024-07-06:
### Is your feature request related to a problem? Please describe

Есть возможность из полетного контролера сделать наземный?

### Describe the solution you'd like

Есть ли решение или готовая кастомная прошивка по полетному контролеру добавить альтернативный мод для наземного дрона?

### Describe alternatives you've considered

Если есть открытый код по управлению двигателей

haslinghuis, 2024-07-06:
Please translate to English.

- Ardupilot provides QGC features out of the box.
- Inav 8 has Express LRS Airport support.

YarosMallorca, 2024-07-06:
@Win3273 keep the discussion on GitHub to English, please.
Betaflight doesn't support ground vehicles. And I'm not aware of any open-source mod that would allow so. Please look into Ardupilot: https://ardupilot.org/

YarosMallorca, 2024-07-06:
@haslinghuis should be marked as off-topic and closed in my humble opinion.

haslinghuis, 2024-07-06:
At the moment we have no plans for a ground control station.

closed 2024-07-06 by haslinghuis