betaflight/betaflight #13754: Onboard ADC does not read voltage on BF4.5 on RushFPV AIO

rsarwar87, 2024-07-06:
### Describe the bug

Hi. I recently BT4.5 on both BetaFPV toothpeak and RushFPV AIO. The BetaFPV, the onboard ADC were able to detect the battery voltage and current draw correctly. However, even on factory settings, the rushFPV FC does not show any battery voltage and current; they are both zero. Rolling back to 4.4.3, I am able to recover the voltage and current readout, even on factory settings. I have taken a dump of both BF4.4 and 4.5, and I see no change in resource allocation.    Onboard ADC is selected on both and resource allocations on both are:

resource ADC_BATT 1 C01
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 C03
resource ADC_EXT 1 NONE

### To Reproduce

Flash both BF4.4 and 4.5 on RushFPV AIO and only the former will have power readouts.

### Expected behavior

onboard ADC should be able to read voltage settings. 

### Support ID

```txt
f0a09f52-b8ba-49e9-9d85-e5d7d50f8854
```

### Flight controller

rushfpv-f7-45a-aio

rsarwar87, 2024-07-06:
aaa, posted too soon, a closer look at the diff between 4.4 (where onboard adc works) and my settings revealed that I am missing this settings:

dma ADC 3 1

it was None on 4.5. setting it to 1 seemed to have restored the functionality.

haslinghuis, 2024-07-06:
@rsarwar87 dma assignment was missing from the configuration.