betaflight/betaflight #13755: "Save and Reboot" not working on MacOs

kouvas, 2024-07-06:
### Describe the bug

macOS Catalina (2012 macbook pro) with latest betaflight configurator 10.10 and NEUTRONRCF435MINI board. 

When I make a change and click save and reboot, nothing happens and i keep waiting for ever. If I click on any other configuration page like failsafe, i get a white screen on the right saying "waiting for data". I have to disconnect the board to be able to use it again. 

Tried from web broswer as well, same issue. Tested on windows and it works just fine, so this must be an issue with macOS only.

### To Reproduce

Use macOS Catalina 10.15.7 and latest betaflight. 
Install latest betaflight and connect to any board. 
Click save and reboot. (nothing happens, no logs)

### Expected behavior

I expect my FC board to be rebooted successfully after i make a change in betaflight and click "Save and Reboot"

### Support ID

```txt
*** Support data submitted *** 
Id: "bd4aaefc-5c3b-445c-8979-bc63292bbba0"
```

### Flight controller

NERC/NEUTRONRCF435MINI(AT32F435G)

haslinghuis, 2024-07-06:
See: https://github.com/betaflight/betaflight-configurator/issues/3514

haslinghuis, 2025-08-10:
Fixed in https://github.com/betaflight/betaflight-configurator/pull/4455

closed 2025-08-10 by haslinghuis