betaflight/betaflight #13757: IRC Tramp and BLITZ Whoop 5.8GHz 2.5W VTX. Incorrect power level control

SergeyBakinskiy, 2024-07-08:
### Describe the bug

When using the VTX table attached to this video transmitter. There is an incorrect control of its power level.

Used powerlevels:

"powerlevels_list": [
            {
                "value": 25,
                "label": "25 "Used powerlevels:

"powerlevels_list": [
            {
                "value": 25,
                "label": "25 "
            },
            {
            },
            {
                "value": 400,
                "label": "400"
            },
            {
                "value": 1000,
                "label": "1W"
            },
            {
                "value": 2500,
                "label": "2W5"

indicate the output rate of the video transmission
Blinking 1 time = 25mW, blinking 2 times = 400mW, blinking 3 times = 1000mW, 
blinking 4 times = 2500mW.

[BLITZ Whoop 5.8G 2.5W VTX(US).json](https://github.com/user-attachments/files/16124925/BLITZ.Whoop.5.8G.2.5W.VTX.US.json)

### To Reproduce

Download this table and switch the power through the configurator.
When the power of 25mw is selected, the LED on the video transmitter flashes 1 time, when the power is selected from 400 to 2.5, it always flashes 4 times,so that the maximum power is used

### Expected behavior

How to edit this table for correct operation of the power level???

### Support ID

```txt
383a2c91-cae5-4126-b3a9-45d3c1c80a93
```

### Flight controller

IFLIGHT_BLITZ_F722

### Other components

VTX BLITZ Whoop 5.8G 2.5W

haslinghuis, 2024-07-08:
- Assume using configurator 10.10
- Is power changing correctly using the power button on the VTX ?
- Is it in PIT mode?
- Tramp protocol used to need having 5 powerlevels.
- Cannot find anything about potential unlock procedure.

[image]

SergeyBakinskiy, 2024-07-09:
1. Yes, using BFc 10.10.0 (c97deaf)
2. Yes, if IRC Tramp disabled in the PORTS tab. All power modes switch from the button to VTX correctly
3. The PIT mod also works correctly. The green LED is on all the time in this mode
4. Then I don't understand why the VTX manufacturer has only 4 power levels and how to make it work.
5. I think there is no need to unlock, all power modes are working

haslinghuis, 2024-07-09:
Somehow provided support ID is incomplete:
```bash
# $X<0timer show
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# 
```

- Please execute `timer show`
- Can you try with https://master.dev.blackbox.betaflight.com/ (needs chrome/chromium - use F12 to expose dev console to check for any errors while changing power level).

Only have SA/MSP based VTX here so running out of options as everything seems configured correctly 
- Ask on [Betaflight Discord](https://discord.betaflight.com/invite) or manufacturer
- Not direct related: with analog build it's recommended to leave out `OSD_HD` in firmware flash options
- Also provide feedback when a solution is found.

SergeyBakinskiy, 2024-07-09:
# Building AutoComplete Cache ... Done!
# 
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

closed 2024-08-17 by github-actions[bot]