betaflight/betaflight #13758: No OSD Tab in Configuratior / No OSD HD showing up after flashing to BF4.5 - Holybro Kakute F7 Mini

x8n1620, 2024-07-09:
### Describe the bug

After flashing my Holybro Kakute F7 Mini to BF4.5 I have no OSD-Tab in the Configurator. OSD is also not showing up in the Goggles.
VTX is a WTFOS Caddx Vista. With BF 4.4.3 the HD OSD is working correctly
setting displayport_msp_fonts = 0,0,0,0 doesn't change anything.

### To Reproduce

Flash a Holybro Kakute F7 Mini to BF4.5
Options: acro trainer, gps, osd (HD)
Wire up a rooted Caddx Vista to UART6

### Expected behavior

OSD is visible in Goggles, Configuration is possible in BF Configurator

### Support ID

```txt
*** Support data submitted *** 
Id: "2ad06d97-3641-4ff9-847b-c6f9c59d868c"
```

### Flight controller

Holybro Kakute F7 Min

### Other components

Caddx Vista rooted with WTFOS on UART6

### Add any other context about the problem that you think might be relevant here

diff all

# version
# Betaflight / STM32F745 (S745) 4.5.0 Jul  6 2024 / 10:15:32 (c155f5830) MSP API: 1.46
# config rev: 4dfe804

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7MINI
manufacturer_id HBRO
mcu_id 0021002d3138511038343530
signature 

# name: DroehniV1.2

# feature
feature -AIRMODE
feature GPS

# serial
serial 0 131073 115200 57600 0 115200
serial 3 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1825 2100 0 0
aux 1 1 2 1825 2100 0 0
aux 2 46 4 1825 2100 0 0
aux 3 13 5 1825 2100 0 0
aux 4 19 1 1300 1700 0 0
aux 5 28 2 900 1175 0 0
aux 6 35 1 1825 2100 0 0
aux 7 47 2 1300 1700 0 0
aux 8 52 1 1300 1700 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxfail
rxfail 0 s 1500
rxfail 1 s 1500
rxfail 2 s 1500
rxfail 3 s 1225
rxfail 4 s 2000
rxfail 7 s 1000

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set acc_calibration = 16,83,-214,1
set rc_smoothing_auto_factor = 250
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set blackbox_sample_rate = 1/16
set blackbox_disable_pids = ON
set blackbox_disable_rc = ON
set blackbox_disable_setpoint = ON
set blackbox_disable_bat = ON
set blackbox_disable_alt = ON
set blackbox_disable_rssi = ON
set blackbox_disable_debug = ON
set blackbox_disable_motors = ON
set blackbox_disable_gps = ON
set dshot_idle_value = 350
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1300
set vbat_max_cell_voltage = 435
set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 360
set ibata_scale = 480
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_min_start_dist = 20
set gps_rescue_ascend_rate = 400
set gps_rescue_ground_speed = 1700
set gps_rescue_descent_dist = 30
set gps_rescue_throttle_hover = 1230
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set report_cell_voltage = ON
set osd_warn_bitmask = 32751
set osd_rssi_pos = 643
set osd_link_quality_pos = 2625
set osd_link_tx_power_pos = 2689
set osd_rssi_dbm_pos = 2659
set osd_rsnr_pos = 644
set osd_tim_1_pos = 3350
set osd_flymode_pos = 2595
set osd_throttle_pos = 2435
set osd_current_pos = 2400
set osd_mah_drawn_pos = 3724
set osd_gps_speed_pos = 2306
set osd_gps_lon_pos = 18464
set osd_gps_lat_pos = 18432
set osd_gps_sats_pos = 2112
set osd_home_dir_pos = 2150
set osd_home_dist_pos = 2144
set osd_flight_dist_pos = 3319
set osd_compass_bar_pos = 122
set osd_altitude_pos = 18626
set osd_warnings_pos = 2679
set osd_avg_cell_voltage_pos = 3691
set osd_battery_usage_pos = 33002
set osd_disarmed_pos = 2393
set osd_esc_tmp_pos = 3095
set osd_flip_arrow_pos = 2460
set osd_efficiency_pos = 2464
set osd_sys_vtx_temp_pos = 263
set osd_stat_bitmask = 6458968
set osd_canvas_width = 60
set osd_canvas_height = 22
set displayport_msp_fonts = 0,0,0,0
set craft_name = DroehniV1.2

profile 0

# profile 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 9

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_expo = 25
set pitch_expo = 25
set roll_srate = 77
set pitch_srate = 77

rateprofile 1

# rateprofile 1
set roll_srate = 27
set pitch_srate = 27
set throttle_limit_type = CLIP
set throttle_limit_percent = 60

rateprofile 2

# rateprofile 2
set roll_srate = 27
set pitch_srate = 27
set throttle_limit_type = CLIP
set throttle_limit_percent = 60

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

haslinghuis, 2024-07-09:
@x8n1620 support ID points to build ID with CORE_BUILD enabled. `CORE_BUILD` does not include features and is for support purposes only.

haslinghuis, 2024-07-09:
Can you enable OSD on configuration tab?

x8n1620, 2024-07-09:
Ouch.... shame on me!
I haven't seen that slider in the configuration tab!!!
After enabling the OSD ... o wonder ... I have a OSD. Soooorry!

closed 2024-07-09 by haslinghuis