betaflight/betaflight #13775: Config strings truncated while use cli dump command through UART port

kedeng, 2024-07-18:
### Describe the bug

Config one of UART port to "Configuration/MSP" and use PC connect it. Send "dump" command  via serial tool, you can see the result string truncated and only last part of strings returned. There is no issue use USB VCP.

### To Reproduce

1. Config one of UART port to "Configuration/MSP"
2. Use serial tool connect this port and send "#"
3. Now in cli mode,  send "dump"
4. Observe result string and compare it with Betaflight Configurator CLI dump string

### Expected behavior

Get full config dump string.

### Support ID

```txt
###

# # Problem description

# # Config strings truncated while use cli dump command through UART port

# ###

# version
# Betaflight / STM32F405 (S405) 4.5.0 Jul 12 2024 / 13:07:04 (f5ba245a7) MSP API: 1.46
# config rev: f5ba245a7
# board: manufacturer_id: SPBE, board_name: SPEEDYBEEF405AIO

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.26V, Core temp=47degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4327, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MSP (53 x 20)
System Uptime: 17 seconds, Current Time: 2024-07-18T04:56:03.782+00:00
CPU:52%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 10
Voltage: 1 * 0.01V (0S battery - INIT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4017 8M
GPS: NOT ENABLED
Arming disable flags: RXLOSS CLI MSP

# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608 JEDEC ID=0x00ef4017
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=0

# dump master

# version
# Betaflight / STM32F405 (S405) 4.5.0 Jul 12 2024 / 13:07:04 (f5ba245a7) MSP API: 1.46
# config rev: f5ba245a7

# start the command batch
batch start

board_name SPEEDYBEEF405AIO
manufacturer_id SPBE

# name: -

# resources
resource BEEPER 1 D15
[... 1201 lines cut ...]
TIM3:
    CH3 : MOTOR 3
    CH4 : MOTOR 4
TIM4:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

### Flight controller

Board: SPBE/SPEEDYBEEF405AIO(STM32F405) BTFL, BTFL version: 4.5.0

### Add any other context about the problem that you think might be relevant here

I made a little change in serial_uart.c,  set uartVTable's writeBuf to NULL and make a custom build. This build seems have no such issue. I guess the reason is cli dump write a lots of data to uart buffer and the buffer have no time to send until the dump process stop, only the tail data send.

ProfDrYoMan, 2025-02-07:
I have the same issue. My USB is gone on my FLWO/FLYWOOF405S_AIO(STM32F405) so I am using UART3 as Config/MSP.

I used this method: https://betaflight.com/docs/wiki/guides/current/broken-usb-rescue

Now I can use the drone and tailor the config, but I am unable to dump the settings and I am not sure if it is save to load a dump and be sure that all got set.

@kedeng what did you do in detail to fix it? Would love to apply myself.

Why is this fix not an option to get into the code?

kedeng, 2025-02-09:
@ProfDrYoMan 
> Why is this fix not an option to get into the code?
I just did a little research, and there may be other side effects. This was only tested in version 4.5.0.

ProfDrYoMan, 2025-02-09:
Setting writeBuf to NULL will disable all write buffering on all serial_uart. I am pretty sure there might be side effect. 

Looking forward to someone finding the root cause and fixing the right way.

ledvinap, 2025-02-10:
uartWriteBuf does not check for buffer overflow.

Can you test #14251 ?

ProfDrYoMan, 2025-02-10:
Is there any way to save my config over UART in the "broken" status right now?
Even stepwise?
I changed so many things already and am afraid need to redo from my head.

ProfDrYoMan, 2025-02-10:
Need to check. But a "diff all" for sure was too much.
But I might be able to do "diff xyz" for all different existing xyz.
Good idea. 

I will check in the evening.
If I manage to save all the settings, I will try the fix.

Is this 4.6beta code or just on top of 4.5.1?
If it is 4.6beta, is this stable enough to keep it on the quad afterwards?

nerdCopter, 2025-02-10:
> Is this 4.6beta code or just on top of 4.5.1? If it is 4.6beta, is this stable enough to keep it on the quad afterwards?

@ProfDrYoMan , The PR is based from in-development 4.6.  Many pilots are flying 4.6 including myself. The flight-code is stable, there may be features that are still a work-in-progress.

Off-subject, there is https://www.expresslrs.org/software/airport/ , but it may be a bit excessive for your needs.

kedeng, 2025-02-11:
> uartWriteBuf does not check for buffer overflow.
> 
> Can you test [#14251](https://github.com/betaflight/betaflight/pull/14251) ?

#14251 can fix this issue.

ProfDrYoMan, 2025-02-11:
> Off-subject, there is https://www.expresslrs.org/software/airport/ , but it may be a bit excessive for your needs.

What should this help me with, @nerdCopter?

nerdCopter, 2025-02-11:
> > Off-subject, there is [expresslrs.org/software/airport](https://www.expresslrs.org/software/airport/) , but it may be a bit excessive for your needs.
> 
> What should this help me with, @nerdCopter?

Broken USB -- CLI over secondary RX via UART. (i never did it, just know that it existed.)

ProfDrYoMan, 2025-02-11:
> Broken USB -- CLI over secondary RX via UART. (i never did it, just know that it existed.)

Ah, I understand, also an option. I just soldered a small connector with wires onto a spare UART of the FC with broken USB and fixed it with cable tie.

haslinghuis, 2025-02-11:
FTDI ?

See https://betaflight.com/docs/development/Serial#serial-port-types

ProfDrYoMan, 2025-02-11:
Yes, that's what I am using on this patched UART.
This is why I ran into this bug when I wanted to save my config.