betaflight/betaflight #13780: BF4.5 Holybro Kakute F4 V2.4 - DShot not working on Motor 4

menschel, 2024-07-23:
### Describe the bug

On a fresh install of BF4.5 on a Kakute F4 V2.4, not all motors are working with dshot.

This seems to be a software issue. There have been similar issues described on various Kakute F4, F7 and even H7 FCs.

### To Reproduce

With Holybro Kakute F4 V2.4 flashed with standard image KakuteF4V2 and dshot capable escs, try to spin motors on motor test page.
Motor 4 is shown as E 100%.

### Expected behavior

All 4 Motors work with dshot.

### Support ID

```txt
# diff

# version
# Betaflight / STM32F405 (S405) 4.5.0 Jun 25 2024 / 17:53:20 (c155f5830) MSP API: 1.46
# config rev: a0cb3dc

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id HBRO

# feature
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 5 64 115200 57600 0 115200

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 13,48,-221,1
set blackbox_sample_rate = 1/16
set blackbox_device = SPIFLASH
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 2

profile 0

rateprofile 0

# end the command batch
batch end

# resource
resource BEEPER 1 C09
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource LED_STRIP 1 C08
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource INVERTER 3 B15
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 B05
resource LED 2 B04
resource LED 3 B06
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_SDI 1 A06
resource SPI_SDI 3 C11
resource SPI_SDO 1 A07
resource SPI_SDO 3 C12
resource ESCSERIAL 1 C07
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C01
resource ADC_CURR 1 C02
resource FLASH_CS 1 B03
resource OSD_CS 1 B14
resource GYRO_EXTI 1 C05
resource GYRO_CS 1 C04
resource USB_DETECT 1 A08

# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin C07 0
# pin C07: DMA2 Stream 2 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
```

### Flight controller

Holybro Kakute F4 V2.4 purchased via banggood

### Other components

Diatone Tyrant S 2017 PDB
4x Holybro Tekko32 F4 Single ESCs flashed to AM32 V2.2
4x Turnigy Viking 2206
HM900 ELRS Receiver

### How are the different components wired up (including port information)

Bare install.
HM900 ELRS Receiver on connected to USART1 (R6 T6)
One ESC + Motor on M1-M4, FC powered via UBat.

### Add any other context about the problem that you think might be relevant here

If one of the other motors is not connected physically, Motor 4 works just fine.

I basically soldered the ESCs one by one and checked if motors are working.

4 only works
1 and 4 both work
1, 3 and 4 all work
In case all motors (1, 2, 3 and 4) are soldered, Motor 4 is not actuated

haslinghuis, 2024-07-23:
- No support ID provided.
- Enable or disable `dshot_bitbang`
-  4.5.1 release should fix this.

menschel, 2024-07-23:
just added support ID
set dshot_bitbang = AUTO

I'll try with
set dshot_burst = OFF
next

menschel, 2024-07-23:
changes in both `dshot_bitbang, dshot_burst` have no effect on my issue

haslinghuis, 2024-07-23:
- What ESC firmware and version is used?
- Try disabling dshot telemetry or led strip

menschel, 2024-07-23:
AM32 with Wraith32_F4 Image Version 2.02

disabling dshot telemetry only removes the visual feedback, Motor 4 does nothing.

disabling led strip shortly kills dshot telemetry on Motor 2 for whatever reason

There also is a funny occurrence that Motor 1 shows twice the temperature then motors 2 and 3.

haslinghuis, 2024-07-23:
Does it work with previous firmware version? 4.4.x / 4.3.2 ?

menschel, 2024-07-23:
latest firmware in 4.4.x / 4.3.x also show this behavior

it basically matches what was described in 

https://github.com/betaflight/betaflight/issues/7366

haslinghuis, 2024-07-23:
According to https://github.com/betaflight/betaflight/issues/7366#issuecomment-469455048 this behavior started with https://github.com/betaflight/betaflight/commit/f8103b8c86082ef05016aafd598f19eb99c9546a

menschel, 2024-07-23:
@kajsilander78 , I've seen you had similar issue on some previous version of Kakute F4 V2.x. Does it work for you with BF4.5 ?

I'm literally two clicks away from purchasing an H7 FC to get rid of this issue.

kajsilander78, 2024-07-24:
Sorry,  have not used any quadcopters for a while. (Dont know what version
I have installed). I think I had some issue with resource mapping in
template?

menschel, 2024-07-25:
> Sorry, have not used any quadcopters for a while. (Dont know what version I have installed). I think I had some issue with resource mapping in template?

Thanks, no problem, I just went plan b and purchased some Matek H743-slim which was too cheap ( 66EUR tax incl.) to miss out.

menschel, 2024-07-25:
just had to edit the original bug report,
the 3rd combination where motor 4 was correctly actuated, was motors 1,2,4.
No clue if it would have worked with 1,3,4 as well.
I'll start pulling out the wires on the jst sh connector and see if I find any rule to that.

menschel, 2024-08-02:
just for info, I have exchanged the Kakute F4 V2.4 against the Matek H743-slim which is running factory betaflight 4.3 and everything works on dshot600

I dislike the Matek FC design though. Connector is on the side instead of front, the standoffs in the kit are 2mm too short, the FC is mounted upside down.

Anyways I'm willing to test any patch that is supposed to fix this issue.

ot0tot, 2024-08-28:
Can you set dshot_burst and dshot_bitbang both to off and test? Please post a new support ID with those settings if you still have issues.

menschel, 2024-08-29:
Hello, the issue is still present.

Motor 4 does nothing, not with and not without bdshot.

new support ID: "a3b09626-82c6-4900-ace5-7dbc4b01830a"

ot0tot, 2024-08-29:
> Hello, the issue is still present.
> 
> Motor 4 does nothing, not with and not without bdshot.
> 
> new support ID: "a3b09626-82c6-4900-ace5-7dbc4b01830a"

You did not disable dshot_bitbang. Please disable dshot bitbang, then test again and submit a support ID if it's not working. 

```
set dshot_bitbang = OFF
save
```

menschel, 2024-08-29:
Sorry, I forgot to save previously.

I see 100% error on motor 4 and 0.11% error on the others now. Motor 4 still does nothing.

ID: "89bfa07c-ef70-4daa-91f5-ca105dfa3d54"

menschel, 2024-08-29:
OK, I am very sorry. I seems I made a fool out of myself.

I started to exchange the jst cable between pdb and fc and apparently I have a partly broken wire in the cable for motor 4.

I realized I never mounted the fc while testing and after mounting it, the wire for motor 4 apparently does connect, so I now get dshot telemetry and control for motor 4 with stock settings.

I did not find it with the H7 because I made a new cable with different pin mappings.

Again, I'm sorry for the noise. Please accept my apologies.

closed 2024-08-29 by menschel

ot0tot, 2024-08-30:
Glad it turned out to be a simple solution. Thank you for the closure.