betaflight/betaflight #13781: RPM Limiter causing uncontrolled RPM overshoot due to excessive filtering phase delay of averageRpm

PMantix, 2024-07-23:
### Is your feature request related to a problem? Please describe

Greetings all,

I've been doing some testing of the RPM Limiter in BF 4.5+ for different motor kVs for the UWL spec racing series (75mm whoop, 31500 RPM limit, ex0802 19,000kV motors, 1s 450mAh cell) and uncovered an issue with the limiter that is creating an uncontrolled overshoot of motor RPM. 

TEST METHOD:

Pretest:
1) Map a switch on transmitter to the following: Switch low --> override throttle to 0%, Switch high --> override throttle to 100%. 
2) Set parameters of experiment in BF configurator, set debug mode to RPM_Limit

Test: 

3) Plug in fully charged battery, place whoop on desk
4) Set throttle to 0%, Arm
5) Holding down whoop with fingers (don't try this at home), sweep throttle from 0-100% a few times to allow limiter to complete initial calibration
6) Toggle throttle override switch to low, hold 1 second, toggle high, hold 1 second, and repeat 3-5 times. This creates a very consistent throttle step input from 0-100% which we can then use to assess the RPM limiter performance. 
7) Disarm, unplug, review blackbox log

EXAMPLE RESULTS:
Here are some example outputs from this kind of testing. 
Plotted vs time are the unfiltered motor RPM from the ESC telemetry, and the filtered averageRpm used by the RPM limiter code
[image]
[image]
[image]
[image]

ISSUE IDENTIFIED:
As you can see in the plots above, the filtered averageRpm lags behind the unfiltered motor RPM significantly. AverageRpm is PT1 filtered using a 6hz cutoff, which by my hand calculation creates about 100ms of phase delay before a step input will be output at 98%. See the figures below with my comments for more details:

[image]
[image]
[image]

So just to reiterate, uncontrolled overshoot is occurring because:
1) The motor response time for my test case from idle RPM -> limit RPM is around 45ms or so
2) The phase delay created by the default averageRpm PT1 filter is around 100ms or so
3) Therefore, the motor RPM is able to quickly exceed the RPM Limit, creating an overshoot, before the limiter is active, since the averageRpm signal will lag behind the actual RPM. 

### Describe the solution you'd like

PROPOSED SOLUTION:
There are several ways to address this, but the easiest is to simply reduce the amount of filtering on averageRpm to reduce the phase delay. From the graph above, you can see a cutoff freq of around 25Hz would reduce the phase delay to something less than the motor response time. I tested this by compiling my own hex's with the averageRpm filter cutoff frequency adjusted from the base 6hz to 2hz, 4hz, 10hz, 15hz, and 25hz. 

An animation of the impact this has to the step response is shown below:
[image]

Most important thing to notice in the animation is 1) As the cutoff frequency goes UP (less filtering), the averageRPM signal more closely follows the unfiltered motor RPM signals - This is good, and 2) By reducing the filter delay to something less than the motor response time we can effectively eliminate any possibility of overshoot, and also 3) notice how dramatic the reduction in response time is for the limiter to begin controlling the RPM (you can see this most easily from the motor command signals in the bottom plot)

So my suggestion is to consider reducing the filtering of the PT1 averageRpm. The appropriate level of filtering will be case specific, and so I think it probably makes sense to allow this to be changed in the CLI, and make those changes apparent in the RPM Limiter OSD output so a spec-agreeable value can be confirmed prior to a race. 

---> Cutoff frequency of PT1 averageRpm should be adjustable from configurator / OSD menu, and that value should be reported on RPM Limiter OSD prearm screen for confirmation of spec

### Describe alternatives you've considered

ALTERNATIVES:
There are problems with my suggestion - for one, reducing the filtering will obviously add more noise. Rather than simply reducing the filtering of the PT1 averageRpm filter, it may be better to also change to a higher order filter to mitigate additional noise. 

Since the phase delay induced by the filtering is static, it may also be possible to account for the phase delay by estimating the non-delayed average RPM from the smoothed averageRpm with knowledge of the phase delay. I've seen this implemented before elsewhere, and while it can be effective - it also tends to amplify high frequency noise in the original signal. So while it may be interesting to consider, I expect there are some technical challenges to overcome there. 

Thanks for reading! Cheers!
-Phil

haslinghuis, 2024-07-23:
@Tdogb

PMantix, 2024-07-26:
Just a note. I've been doing some testing on the Street League RPM limiter as well and found the same situation as above - excessive averageRpm filtering delay leads to RPM overshoot. The SL developer added an adjustable k factor for the PT1 filter of the averageRpm and this worked well to resolve the issue.

Tdogb, 2024-08-03:
Hey, thanks for you in depth analysis! I chose to use a relatively high level of filtering to ensure that people won't have issues with the rpm limiter adding noise to the motors. I believe only 50ms of overshoot is not bad. If you can prove that it is possible to abuse this to gain an advantage in a race then we could consider changing it.

PMantix, 2024-08-03:
> Hey, thanks for you in depth analysis! I chose to use a relatively high level of filtering to ensure that people won't have issues with the rpm limiter adding noise to the motors. I believe only [...]

Thanks for taking a look! 

I agree 50ms of overshoot is not too bad, but it’s enough to allow several thousand RPM of overshoot difference between motors with large differences in kV. It’s enough to make a noticeable impact to acceleration, especially when transitioning from low to high throttle repeatedly through obstacles. Honestly, I argued that even that difference doesn’t make an impact in a race, but regardless it caused the UWL spec to ban all motors except the EX0802 from HM. So my motivation with this suggested change was less about preventing people from taking advantage of the limiter, and more about improving the performance of the limiter so more motors with larger differences will perform more similarly than they do now. (Which is really “same difference”, but just trying to give some more background).

Also, I just want to point out that allowing the Average RPM filter cutoff to be adjustable doesn’t have any impact to motor noise if the default settings remain as they are now. The pull request I made just gives more flexibility to make adjustments for specs that might need more or less filtering. Tuning flexibility is a good thing I think when it comes to finding balance between performance and noise mitigation, right?

And this is a little tangential, but if the main concern is the potential for increasing motor noise…  then I actually have another branch I’m working on that changes the filtering strategy of the RPM limiter so that the motor noise is reduced even when decreasing the average RPM filtering. As most of the motor noise is coming from taking the numerical derivative for D term, I simply put a PT1 filter on D term, ha. Seems to work great! 😄

PMantix, 2024-08-03:
Here's a comparison between the RPM limiter response to a step input in throttle at t = 0s. Same motors, same battery, same level of charge. Differences are 1) I've increased the AverageRPM filter cutoff from 6 -> 350, 2) I've added a D term PT1 filter and set its cutoff to 350, and 3) I've adjusted the PIDs from 25, 10, 8, to 10, 10, 1, respectively.

[image]

The filter delay is closer to 100ms, not 50ms. It's the motor response time that is 50ms (that's what the "Average Filter Delay Characteristic" figure above tried to point out). 

That large delay compared to the motor response time allows the motors to overshoot the limit for ~200ms, and by several thousand RPM (the more important problem, I think). This is quite significant in my opinion. You can easily feel the difference, hear the difference, and see the difference in acceleration when pumping the throttle in a hover. 

The changes I made in comparison ends up having much less overshoot, and qualitatively "feels" better and sounds better. It does seem to be noticeably less punchy when pumping the throttle in a hover, but I haven't attempted to measure that quantitatively. 

Also note there isn't a noticeable difference in motor noise between the two. D term filtering FTW! 😄

closed 2024-09-10 by github-actions[bot]