betaflight/betaflight #13782: HAKRCF722D does freeze in flight with DJI O3 and SUMD/HoTT-Telemetry

Ralfde, 2024-07-24:
### Describe the bug

The copter does randomly disarm in flight and it looks like the flight control is frozen. Unplug and connect it again does make it working again. The hardware must be okay, because it does not happen with inav 7.1.2. The problem does occur with 4.4.3 and 4.5.
I have GPS connected and DJI O3 with OSD and I use SUMD with HoTT-Telemetry.
I have the impression, that these freeze do also happen on the table, if the Betaflight Configurator is connected. 

### To Reproduce

fly very low and slow around to protect your copter and just wait until it randomly fall down.

### Expected behavior

the copter should just not fall down

### Support ID

```txt
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Jun 15 2024 / 12:22:49 (c155f5830) MSP API: 1.46
# config rev: b98adf6

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HAKRCF722D
manufacturer_id HARC
mcu_id 004e003d3232511930383830
signature 

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# serial
serial 1 64 115200 57600 0 115200
serial 2 2 115200 115200 0 115200
serial 3 8 115200 57600 19200 115200
serial 5 131073 115200 57600 0 115200

# beacon
beacon RX_LOST

# map
map TAER1234

# led
led 0 0,0::G:2
led 1 1,0::C:2

# mode_color
mode_color 6 5 10
mode_color 6 7 2

# aux
aux 0 0 3 1400 2100 0 0
aux 1 1 0 975 1300 0 0
aux 2 2 0 1400 1600 0 0
aux 3 6 2 1800 2000 0 0
aux 4 27 1 1800 2025 0 0
aux 5 46 1 950 1200 0 0
aux 6 35 3 1975 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 0 h
rxfail 1 h
rxfail 2 h
rxfail 3 s 1400
rxfail 4 s 1100
rxfail 7 h
rxfail 11 h
rxfail 15 s 1000

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 600
set dyn_notch_count = 2
set dyn_notch_q = 400
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_lpf_hz = 10
set acc_calibration = 21,-9,35,1
set rssi_channel = 16
set fpv_mix_degrees = 40
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_poles = 12
set failsafe_delay = 20
set failsafe_throttle_low_delay = 200
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set ibata_scale = 200
set yaw_motors_reversed = ON
set small_angle = 35
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_alt_mode = CURRENT_ALT
set gps_rescue_return_alt = 24
set gps_rescue_ground_speed = 800
set gps_rescue_throttle_hover = 1350
set gps_rescue_min_sats = 7
set simplified_gyro_filter_multiplier = 120
set osd_vbat_pos = 14369
set osd_rssi_pos = 14392
set osd_flymode_pos = 14701
set osd_crosshairs_pos = 14573
set osd_ah_sbar_pos = 14574
set osd_ah_pos = 14446
set osd_current_pos = 14376
set osd_mah_drawn_pos = 14384
set osd_gps_speed_pos = 14615
set osd_gps_lon_pos = 14417
set osd_gps_lat_pos = 14401
set osd_gps_sats_pos = 14552
set osd_home_dir_pos = 14638
set osd_home_dist_pos = 14679
set osd_compass_bar_pos = 14666
set osd_altitude_pos = 30871
set osd_warnings_pos = 14729
set osd_disarmed_pos = 395
set osd_flip_arrow_pos = 14745
set osd_stat_bitmask = 16188
set osd_canvas_width = 30
set osd_canvas_height = 13
set vtx_band = 5
set vtx_power = 2
set vtx_low_power_disarm = ON
set rpm_filter_q = 1000
set rpm_filter_fade_range_hz = 100

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_static_hz = 105
set dterm_lpf2_static_hz = 210
set anti_gravity_gain = 100
set p_pitch = 59
set i_pitch = 79
set d_pitch = 30
set f_pitch = 129
set p_roll = 49
set i_roll = 66
set d_roll = 27
set f_roll = 108
set p_yaw = 49
set i_yaw = 66
set f_yaw = 108
set d_min_roll = 27
set d_min_pitch = 30
set simplified_i_gain = 75
set simplified_d_gain = 90
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 90
set simplified_pitch_pi_gain = 115
set simplified_dterm_filter_multiplier = 140

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 1

# save configuration
save
#
```

### Flight controller

HAKRCF722D

### Other components

DJI O3 UART6
SUMD on RX2
HoTT-Telmetry on TX4
GPS UART3

### How are the different components wired up (including port information)

DJI O3 UART6
SUMD on RX2
HoTT-Telmetry on TX4
GPS UART3

haslinghuis, 2024-07-24:
To further narrow down the issue:

- provide support ID
- did it fail on firmware 4.3.2 / 4.4.x
- what specific GPS is used?
- temporary disable telemetry
- provide blackbox
- lower cpu load by using dshor 300 / 4K pid loop

Ralfde, 2024-07-24:
provide support ID What is it?
did it fail on firmware 4.3.2 / 4.4.x It failed also with 4.4.3
Did 4.3.2 support already DJI O3 OSD? If yes, I could try.
what specific GPS is used? Mateksys M10, worked with CRSF
temporary disable telemetry: I also think it could be a problem of HoTT-Telemetry, maybe in combination with DJI O3 OSD.
How could I make own builds with SUMD and HoTT-Telemetry? In which files could I find the #defines? 

I would like to try a change of the following file, which made the Telemetry working faster and more stable with inav. But at inav it made not trouble like freezing before.
[hott.zip](https://github.com/user-attachments/files/16361018/hott.zip)

Ralfde, 2024-07-24:
I tested more on the PC and it looks like the freeze can be seen here, too. So flying is maybe not needed in the first step.
With DShot 300 it freezes also, just takes longer.
With DShot 150 it is running now for at least 15min, but I will test longer.
If I switch off HoTT-Telemetry, DShot600 was running over 1 hour without problems.
Could the problem be a combination of HoTT-Telemetry and DShot?
If somebody would have a debug possibility and HoTT, it should be much more easy to find the problem.

haslinghuis, 2024-07-24:
Afaik nobody of our team uses HOTT telemetry. Please test with 4.3.2 / 4.4.0 and up to see where it freezes.

Ralfde, 2024-07-25:
It looks like the 4.4.0 is okay and the 4.4.2 does not work any more.
What I recognized, too, with 4.4.2 the GPS active symbol on the Configurator is shown much later as functional compare to 4.4.0.

Ralfde, 2024-07-25:
At the Moment it looks like the branch a0d009 from 14.04.2023 is still okay
and 
20747 from 14.04.2023 is not okay any more

Ralfde, 2024-07-25:
For any reasons the GPS is also detected much slower in the versions after 4.4.0.

haslinghuis, 2024-07-25:
@SteveCEvans hott telemetry freezes flight controller after a certain time after https://github.com/betaflight/betaflight/commit/20747ec3e37b4e7b52cb874805c1ba9ab05948d8.

AFintheair, 2024-07-27:
I've noticed similar behavior with a HAKRC F722D and BF 4.5.0. Several times in slow, low flight, the connection from the O3 to the goggles was suddenly disconnected at a distance of about 10m to 50m. Since I was low, I immediately disarmed. In addition, the ESC very often rebooted directly when landing (during the disarm?). I always arm just above the ground.
I checked all electric contacts for sure.
However, I use Crossfire 50Hz with telemetrie!  Unfortunatelly, I have changed FC in the meantime and can not do more tests. I used an Matek 5883 GPS, Compass enabled but not used for GPS Rescue maybe this info will help.

Ralfde, 2024-10-03:
Running 4.5.2 on the table, the FC did still freeze after some time.

Ralfde, 2024-10-09:
The FC did still freeze after some time, when using HoTT-telemetry with PID loop time 8kHz and DShot600. 
[Alpha_150_BF_4_6_0_BTFL_cli_HAKRCF722D.txt](https://github.com/user-attachments/files/17303943/Alpha_150_BF_4_6_0_BTFL_cli_HAKRCF722D.txt)
I'm testing now with 2kHz and DShot300 and will report again. But last time it didn't make a difference, even if I changed to Multishot.

If you don't have a HoTT receiver, you may use a ELRS PWM receiver with ESP32 and adjust SUMD and HoTT-telemetry at ELRS 3.5.0.

Ralfde, 2024-10-09:
As expected it did freeze now also with 2kHz PID loop time and DShot300.
It doesn't freeze, if I switch off the HoTT-telemetry.

closed 2026-09-26 by blckmn