betaflight/betaflight #13783: Incorrect rateprofile switch behavior

IRBaboon595, 2024-07-26:
### Describe the bug

The bug was accidentally found during angle limits checking in Betaflight Master branch.
1. I have build hex for SpeedyBeeF405V3 and HAKRCF405V2 and results are the same;
2. What I have encountered - I have preconfigured first rateprofile as exponential func, second and third as linear.
Angle limit - 45 degrees. I got this 45 limits with picth/roll at rateprofile 1, but got half of limit with rateprofile 2 and 3;
Next - a gave rateprofile 3 the same parameters as in the first (expo). This resulted in correct limit angles for 1 and 3 RP and incorrect (nearly half ~22 degrees) for 2 RP. Than I made all of RPs to be exp and than linear - in both cases I got CORRECT limit angles.
3. To make sure that I was still сonscious - I made the same test for 4.3.2. All correct.

### To Reproduce

1. Build Betaflight master for SpeedyBeeF405V3;
2. Flash the FlightController;
3. Configure with BF Configurator for your remote;
4. Configure three rateprofiles as described in bug_description.
for example center sensivity = 65, max rate = 680, expo = 0 (for rateprofile 1 - expo) AND center sensivity = 360, max rate = 360, expo = 0 for pitch/roll and center sensivity = 270, max rate = 270, expo = 0 for yaw (for rateprofiles 2 and 3 - linear);
5. Configure angle limits for desired value;
6. Enable adjustments two switch between rateprofiles;
7. In modes enable Angle mode to be enabled with arm;
8. test angle limits for pitch/roll from (5).

### Expected behavior

I do expect that when I switch between preconfigured Rateprofiles - I will get correct angle limits at MAX stick position.
As described in bug_description - if I configure rate profiles as one is exp and others are lin - for lin I do get INCORRECT limit angles.

### Support ID

```txt
64d0cb3e-9edf-409f-9c90-c9b8a95ff2b4
```

### Flight controller

SpeedyBeeF405V3

haslinghuis, 2024-07-26:
support ID only shows rate profile 2 ?

IRBaboon595, 2024-07-26:
> support ID only shows rate profile 2 ?

seen it - sorry seems I just do not know much but as I recon in support ID it IS showing the current chosen rateprofile. So as it was by default chosen as rateprofile 3 - u see "2", I have just changed it to rateprofile 1 and now I see "0" there.

haslinghuis, 2024-07-26:
Internal index is 0 based while user index is 1 based. Was assuming support ID would provide the other rate profiles too.

Please provide output of `diff all`

_Use three backticks ( ` ) before and after the output on a separate line._

IRBaboon595, 2024-07-26:
```
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.3.2 Nov 28 2022 / 07:26:30 (60c9521) MSP API: 1.44

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPEEDYBEEF405V3
manufacturer_id SPBE
mcu_id 002b00223133471234333035
signature 

# feature
feature -RX_PARALLEL_PWM

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 0 900 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

# rateprofile 1
set roll_rc_rate = 36
set pitch_rc_rate = 36
set yaw_rc_rate = 27
set roll_srate = 36
set pitch_srate = 36
set yaw_srate = 27

rateprofile 2

# rateprofile 2
set roll_rc_rate = 36
set pitch_rc_rate = 36
set yaw_rc_rate = 27
set roll_srate = 36
set pitch_srate = 36
set yaw_srate = 27

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
# 
```

Here is the output

IRBaboon595, 2024-07-26:
The same thing recorded for 4.5.0 - release, 4.5-maintanance.

closed 2024-09-05 by github-actions[bot]

nerdCopter, 2024-09-05:
- https://github.com/betaflight/betaflight/pull/13862
- https://github.com/betaflight/betaflight/pull/13864