betaflight/betaflight #13790: Yaw I term and pidsum increasing causing  disarm throttle runaway mid flight

Star4fish, 2024-07-28:
### Describe the bug

Sometimes but not everytime applying yaw moves causes yaw I term ,pidsum and yaw pid error to build up what leads motors 2 and 3 uncontrollable speedup to 90% with 0% throttle, hover and flyaway without any throttle control and disarm mid flight with throttle runaway warning that I've read should be disable after proper takeoff. I'm not sure it is #13486 or #13506

Video
[https://drive.google.com/file/d/1eVYu3s-NW85bOAVnCV9U24s5-uFu_0xb/view?usp=drive_link](url)

BB Log
[https://drive.google.com/file/d/1IFozjSNfw8QWmX4rxh2_15uvQnC0zaJR/view?usp=drive_link](url)

[image]
[image]
[image]
[image]

### To Reproduce

Tested on BF 4.5, Blujay 0.19 and 0.21, props in and out, different props

It happend only when yaw stick move is applied, pitch and roll never causing it.

### Expected behavior

Don't cause hover without throttle input, flayaway and disarm mid flight

### Support ID

```txt
1aef7cf4-6402-4212-9fa1-de1963f8edd0
```

### Flight controller

STACK SPEEDYBEE F405 V4 , BLS 60A

### Other components

Frame Speedybee Bee 35 Pro  
Motors Iflight XING2 2205 2300KV
RX Speedybee Nano 2.4GHz ELRS
VTX Speedybee TX Ultra 1600
Cam Caddx Ratel Pro
GPS Foxeer M10Q-120
Buzzer Iflight YR50B_S

SupaflyFPV, 2024-07-28:
definitely something weird going on here....not totally sure might have to call in the Iterm expert...@ctzsnooze

[image]

SupaflyFPV, 2024-07-28:
Are you flying in windy weather?

Star4fish, 2024-07-28:
> Are you flying in windy weather?

no wind, weather was fine

Star4fish, 2024-07-28:
> definitely something weird going on here....not totally sure might have to call in the Iterm expert...@ctzsnooze
> 
> [image]
> 

you have set graph for pid error ROLL which is ok, but pid error YAW shows a lot of errors 🫤

SupaflyFPV, 2024-07-28:
> > definitely something weird going on here....not totally sure might have to call in the Iterm expert...@ctzsnooze
> > <img alt="Screenshot 2024-07-28 at 12 40 21" width="1684" src="https://private-user-images.githubusercontent.com/46397720/352809438-8c87547a-3bcb-4a97-ab6f-05acce760dbb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIxNzEyMDMsIm5iZiI6MTcyMjE3MDkwMywicGF0aCI6Ii80NjM5NzcyMC8zNTI4MDk0MzgtOGM4NzU0N2EtM2JjYi00YTk3LWFiNmYtMDVhY2NlNzYwZGJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzI4VDEyNDgyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk1ZjBmOGQ2OTI4M2NmZDAwODRlMGIxZDM1M2YzNTdlMWQwMjFlN2MyZTA0YmU3NWU4MDk3MzRkZWJiMTUxNDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zGjH62KLmy20cCSCwTqU9UJElfMlpdd7mZZLDh_CpeU">
> 
> you have set graph for pid error ROLL which is ok, but pid error YAW shows a lot of errors 🫤

thank you - thats more like it. Picked Roll by mistake. What is strange is that there is initially no PID error. It suggests that the quad is reacting to your input. Anyway - the Iterm goes way higher so maybe you are tracking its action...?

[image]

Star4fish, 2024-07-28:
> thank you - thats more like it. Picked Roll by mistake. What is strange is that there is initially no PID error. It suggests that the quad is reacting to your input. Anyway - the Iterm goes way [...]

yes its mostly flying well but after 2 or 3 pack that happend and it always happend when I do yaw turn, during that it responds to RPY, sometimes starts wiggling but throttle is out of control, cannot descend. Nearly all of the times if that happend I could regain control of throttle after a few seconds and safely land but total flyaway and disarm like that happend three times now

Star4fish, 2024-07-28:
That was the other time but I've manage to recover it from crash, I term on yaw was going crazy too... but after some sec I've regain throttle control.

BB log 
https://drive.google.com/file/d/1d5fTYRL0-G3_FCCXH0LEyfI1pYOGNKm2/view?usp=drive_link

Vid with BB log synced
https://drive.google.com/file/d/17-Vy0i05hfJNrpS5YAOVW7jqaz5DA2GX/view?usp=drive_link

ctzsnooze, 2024-08-11:
The reason is that you are not flying with zero yaw input.  You are flying with a small but persistent yaw demand for many seconds (5 seconds here).  The quad is trying to keep yawing at the rate you intend, and it accumulates iTerm on yaw to achieve this.  
Fast yaw change comes from motor momentum, ie from the acceleration of two motors and two opposing motors; their reverse torque turns the quad.  But this cannot keep happening over time, we can't accelerate motors forever.
For sustained yaw change the quad has only the air drag on the props.
Some props have more, and some less, relative to frame mass and other drag elements, and some quads that are flown forwards but at an angle had a rotational drag as well.
The only way the quad has, to achieve ongoing yaw, or to oppose an ongoing yaw drag from flying 'crabwise' or with an offset drag element, is to keep two motors going faster than another pair.  The driver for this is iTerm; it is always iTerm that accumulates to deal with small persistent errors.
This image shows yaw setpoint and gyro.  There is 3-4s of sustained yaw, leading iTerm to accumulate to its max of 400.  This causes one pair of motors to be at max and the other at min.  The two at max generate thrust and it keeps climbing.
[image]
So:
- check the build for symmetry (no drag that might tend to yaw it). The worst ad large goPros way up front which, in forward flight, tend to drag the quad hard around when a small amount of yaw is requested, so that the yaw system has to oppose a rotational drag all the time
- stop flying with small amounts of yaw input for long periods of time (fly straight, or use yaw briefly to turn)
- use iTerm windup at say 50.  This will reduce the amount of iTerm accumulation when iTerm exceeds 50.  It won't stop it happening but will make it slower to accumulate.
- consider reducing the PID sum limit for yaw to say 250.  That will limit how much this iTerm saturation can achieve.  Unfortunately iTerm will grow inside the limit and will be slow to go away.

Overall its best to fly straight :-)

A better solution is in my PR https://github.com/betaflight/betaflight/pull/13543

This allows the user to set a separate pidSum limit for yaw, and then stops iTerm from exceeding the itermWindup percentage of that, and does it independently for each axis.  With this PR you can set the pidSumLimit for yaw to say 250 and iTermWindup to 50% and then iTerm on yaw can't exceed 125 - whereas now it maxes at 400.

ctzsnooze, 2024-08-11:
Here you've been flying almost 8s with sustained yaw, leading to pid iTerm on yaw reaching its max at 40%.  Unfortunately with our current code that limit of 40% is on the pid Sum, and the only thing limiting iTerm is iterm_windup, which is coded so that you only get zero gain in iTerm when the motor differential is 85%.  So both things allow significant iTerm windup.
Then it seems that once you realise that the quad is climbing from all the iTerm, you strongly yaw the other way and drop throttle.  
This starts to do the 'right thing' but you push too hard and not for long enough.
It took 8s to accumulate iTerm, it will take time to get rid of it; the solution is to use sustained yaw the other way.

Please try the PR mentioned above, but be sure to set iTermWindup to 50 and reduce the pidsumlimit for yaw to say 250.

Or just fly straight and be mindful of not holding yaw for extended periods.

ctzsnooze, 2024-08-11:
Sorry the image for above comment
[image]

ctzsnooze, 2024-08-11:
What props are you using?
Please share a photo of the quad, looking at it from the perspective of drag elements and frontal elements.
Please try the PR mentioned above with the settings above and make a log to see what happens if you do the same long duration yaw inputs with the newer coding.

Star4fish, 2024-08-12:
During that time I've made some tests and I restored the previous betaflight 4.4.3 that this FC comes in, with the same settings as I set up in 4.5 aparat from prop direction (I left it default to props in) and all problems disappeard, it flies like a charm.  
After that I changed action 2 tpu holder from one that have hinge on the back to hinge on the botton and dismantled camera soft mount to lower the position of camera, because it constantly damaged camera screen :/

I think it's not possible to fly with 0 yaw input, you always have to correct something ;) But it mostly happend when I performed a sharp 180 turn. 

Sometimes happened even in super straight flights, I've lowered the throttle to 0, took my hands away from RC and it was staying on the same altitude or increasing altitude to the time it stops and slowly starts to descend. But I have no BB log from that. 

Bee35 before change camera tpu mount
[image]
[image]

Props - Gemfan D90-3

What about that disarm midflight with throttle runaway warning? Isn't that be disable after crossing 20% throttle and proper takeoff?

ctzsnooze, 2024-08-30:
Forward mounting of large objects causes all kinds of issues in forward flight on low authority builds in particular.  

They act like an aerodynamic 'fin' or 'sail' but at the wrong end of the craft.  If you are keeping altitude constant and pitch constant and do a sustained yaw at speed, the airflow makes it very difficult for yaw iTerm to not build up to handle this aerodynamic problem.  There are other problems, in that the front pair of props are loaded much more heavily than the rears, so that it won't handle well, and when you yaw and request a diagonal pair of motors to speed up, you'll get problems with pitch and roll offsets at the same time.  For this kind of build you want a small camera, not a huge one, and you want it as far back as possible.  And put the battery as far back as possible to balance the weight and aerodynamics.

Please test https://github.com/betaflight/betaflight/pull/13543

Set iTerm_windup to 50% and `possum_limit_yaw` to say 250 and it should solve the original issue.  Please try that out.

closed 2026-09-26 by blckmn