betaflight/betaflight #13792: Smartaudio don't work with uart1 tx of Betafpv 1s 5A AIO

useisbb, 2024-07-28:
### Describe the bug

I monitor to UART1 TX without any data when T1 was configured as Smartaudio, but when changed to IRC Tramp kept receiving some data, but I didn't confirm if it was correct or not!

### To Reproduce

Install the firmware 4.4.3 with betaflight configuartor, and loaded my config with CLI
[BTFL_图传能显示但是不发任何数据.txt](https://github.com/user-attachments/files/16403829/BTFL_.txt)

### Expected behavior

I Expected the Data of SmartAudio is sent on my board uart1 TX 

### Support ID

```txt
# 

# dump

# version
# Betaflight / STM32F411 (S411) 4.4.3 Jun 25 2024 / 14:00:40 (738127e7e) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# name: ack75

# resources
resource BEEPER 1 A14
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 A08
resource MOTOR 6 A10
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A08
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
[... 947 lines cut ...]
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

#
```

### Flight controller

betafpv 1S 5A aio

### Other components

Firmware version is 4.4.3
[image]

rpisarew, 2024-07-31:
I have the same issue with the meteor75 Pro, it has the same board.

It works for me with the provided [firmware (BF411 4.4.0)](https://support.betafpv.com/hc/article_attachments/16846211844889) by Batafpv, but not with the latest.

haslinghuis, 2024-07-31:
SA works on 2S version of the board with 4.5 and 4.6 (using soft serial).
Should SA be connected to TX2 on the [1S](https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-5a-aio-brushless-flight-controller-elrs-2-4g
) version?

[image]

rpisarew, 2024-07-31:
I my case it is connected to tx2.

haslinghuis, 2024-07-31:
Flashing BETAFPVF411 should work. Please provide a support ID from CLI tab. Do not use a dump to restore settings.

rpisarew, 2024-07-31:
> Flashing BETAFPVF411 should work. Please provide a support ID from CLI tab. Do not use a dump to restore settings.

I don't get it. I know that the flashing works, I can switch back and forth without any issues.
My issue is that the SA works for me only on the older version (Betaflight / STM32F411 (S411) 4.4.0 Feb 14 2023 / 05:25:45 (4605309d8) MSP API: 1.45) and not on the 4.5.

I didn't use an CLI dump after flashing 4.5. I configured everything from scratch.

haslinghuis, 2024-08-01:
@rpisarew no need for another issue. Please provide support ID for working and non-working version. Also try 4.5.1 / 4.6.0-dev ?

rpisarew, 2024-08-01:
> @rpisarew no need for another issue. Please provide support ID for working and non-working version. Also try 4.5.1 / 4.6.0-dev ?

No success with 4.5.1 and 4.6.0-dev.

Support ID (created on 4.6.0-dev): 24c20743-9e2d-419c-8c36-bc3e55c3c79c

haslinghuis, 2024-08-01:
1. Working configuration has `dshot_bitbang` enabled (gyro has no dma lock).
2. Non working configuration has `dshot_bidir` disabled.

rpisarew, 2024-08-01:
> 1. Working configuration has `dshot_bitbang` enabled (gyro has no dma lock).
> 
>     2. Non working configuration has `dshot_bidir` disabled.

I try to enable it in the evening and report you back.

haslinghuis, 2024-08-01:
Possible resource conflict (would have to check STM32 tables)
No dma lock means it has to use interrupts instead of direct memory access.

First try enabling `dshot_bidir` and `dshot_bitbang` one by one.

rpisarew, 2024-08-01:
I tried all permutations (on/off) of this two properties, unfortunately it didn't helped.

haslinghuis, 2024-08-01:
Please check 4.4.2 still works and 4.4.3 does not?

rpisarew, 2024-08-01:
I works with 4.4.2 and 4.4.3.

haslinghuis, 2024-08-01:
You can test these builds:

[betaflight_4.6.0_STM32F411_BETAFPVF411_without_13715.zip](https://github.com/user-attachments/files/16459499/betaflight_4.6.0_STM32F411_BETAFPVF411_without_13715.zip)
[betaflight_4.6.0_STM32F411_BETAFPVF411_without_13764.zip](https://github.com/user-attachments/files/16459500/betaflight_4.6.0_STM32F411_BETAFPVF411_without_13764.zip)

rpisarew, 2024-08-01:
`betaflight_4.6.0_STM32F411_BETAFPVF411_without_13715.zip` is working and `betaflight_4.6.0_STM32F411_BETAFPVF411_without_13764.zip` is not.

BTW: How can I trigger the GitHub Action for a build on my own fork?

haslinghuis, 2024-08-01:
This build is without both PR's to see if PR 13764 is to blame or PR 13715:

[betaflight_4.6.0_STM32F411_BETAFPVF411_without_both.zip](https://github.com/user-attachments/files/16459664/betaflight_4.6.0_STM32F411_BETAFPVF411_without_both.zip)

rpisarew, 2024-08-01:
> This build is without both PR's to see if PR 13764 is to blame or PR 13715:
> 
> [betaflight_4.6.0_STM32F411_BETAFPVF411_without_both.zip](https://github.com/user-attachments/files/16459664/betaflight_4.6.0_STM32F411_BETAFPVF411_without_both.zip)

For me it looks like this [PR](https://github.com/betaflight/betaflight/commit/ec8c59b49b72f70dca9594b281d51d7135ac2091) has the issue with the `SERIAL_BIDIR_NOPULL` flag.

This build works.

haslinghuis, 2024-08-01:
Please check this build: [betaflight_4.6.0_STM32F411_BETAFPVF411.zip](https://github.com/user-attachments/files/16459845/betaflight_4.6.0_STM32F411_BETAFPVF411.zip)

rpisarew, 2024-08-01:
> Please check this build: [betaflight_4.6.0_STM32F411_BETAFPVF411.zip](https://github.com/user-attachments/files/16459845/betaflight_4.6.0_STM32F411_BETAFPVF411.zip)

This one is not working.

haslinghuis, 2024-08-01:
This one works?

[betaflight_4.6.0_STM32F411_BETAFPVF411.zip](https://github.com/user-attachments/files/16459955/betaflight_4.6.0_STM32F411_BETAFPVF411.zip)

rpisarew, 2024-08-01:
With your latest [PR#13797](https://github.com/betaflight/betaflight/pull/13797) it is not working.

haslinghuis, 2024-08-01:
Temporary removed that part now. And should only apply when the port has softserial indentifier. So not sure why that does not work. Now it basically reverts PR 13764

rpisarew, 2024-08-01:
> Temporary removed that part now. And should only apply when the port has softserial indentifier. So not sure why that does not work. Now it basically reverts PR 13764

hmm, it is still not working any more.

haslinghuis, 2024-08-01:
Same as PR but build locally:

[betaflight_4.6.0_STM32F411_BETAFPVF411_PR_LOCAL.zip](https://github.com/user-attachments/files/16461213/betaflight_4.6.0_STM32F411_BETAFPVF411_PR_LOCAL.zip)

rpisarew, 2024-08-01:
> Same as PR but build locally:
> 
> [betaflight_4.6.0_STM32F411_BETAFPVF411_PR_LOCAL.zip](https://github.com/user-attachments/files/16461213/betaflight_4.6.0_STM32F411_BETAFPVF411_PR_LOCAL.zip)

still not working.

haslinghuis, 2024-08-01:
> betaflight_4.6.0_STM32F411_BETAFPVF411_without_13764.zip is working and betaflight_4.6.0_STM32F411_BETAFPVF411_without_13715.zip are not.

Guess we reverted the wrong PR. Please check as PR got updated.

rpisarew, 2024-08-01:
Now it's working again :)

Thank you very much for your support and time!

ledvinap, 2024-08-02:
@haslinghuis : This is crazy. It seems like different options are switched on and off to match different smartaudio versions.

Smartaudio noninverted UART and 0V as idle level is begging for problems.

We should only handle bug reports with oscilloscope/LA trace and handle only mismatch with TBS SA documentation

haslinghuis, 2024-08-02:
@ledvinap the "fix" only reverts STM32 changes made in  [ee9172f](https://github.com/betaflight/betaflight/commit/ee9172f4da9aa6f4edfa11babe9f2f7cfc180a32)

ledvinap, 2024-08-02:
But what is wrong with the changes? The idea was to make different chips honor serial setting (behave in more unified way). 
It it does break smartaudio, it means that SA options are wrong and/or that SA does not work on some chips.

Why do we have `SERIAL_BIDIR_PP_PD` if it does not set pulldown? Either it is useless option and can be removed from code, or it should not be set when if breaks some protocol.

haslinghuis, 2024-08-02:
This code was updated before by @jflyper in 2017: fdfe9e8af355ee70edecabe206536ad95fc69b6e.
Is there any documentation about SA options?

bkleiner, 2024-08-02:
So I am confused - which configuration is working now? No pulldown, no dummy byte?

haslinghuis, 2024-08-08:
@rpisarew please test #13585

rpisarew, 2024-08-08:
> @rpisarew please test #13585

I can do that on saturday.

rpisarew, 2024-08-10:
> @rpisarew please test #13585

SA is not working with this build.

useisbb, 2024-08-31:
> SA works on 2S version of the board with 4.5 and 4.6 (using soft serial). Should SA be connected to TX2 on the [...]

The T2 is broken on my board, so I want to change to T1

useisbb, 2024-08-31:
Thank you。 This bug can be avoided by the old version as 4.4.0 or 4.4.3

haslinghuis, 2024-08-31:
Reopening as we need to fix this for current firmware.

PR #13797 (2024-10-24): Fix SmartAudio (STM32F4)
- fixes #13792
- requires adding custom define `SMARTAUDIO_NOPULLDOWN`
closed 2024-10-24 by blckmn