betaflight/betaflight #13793: Controlled descent while disarmed

stubb0rnCoder, 2024-07-29:
### Is your feature request related to a problem? Please describe

A few days ago, while flying, I had a situation where I had to disarm the copter mid-air, about 30m (10-ish feet) above the ground. While watching the video feed I could see that the copter was drifting towards and beyond a wall which would make recovery extremely complicated. The time it took me to assess the quads state and what prevented me from re-arming the copter was already behind said wall and video signal was lost.

### Describe the solution you'd like

Changing the "disarmed" mode to a "limited disarm" where the PID loop and throttle are deactivated but inputs to pitch, roll, yaw still allow for e.g. 1-20% of motor throttle which would allow to somewhat control a descent and "glide" towards a safer landing zone.

Setting the max. motor throttle signal to zero would effectively be what is "disarmed" today.

### Describe alternatives you've considered

Instead of disarming, switching to a tuned-down PID and rate profile would come closest to what I have in mind but this would still have airmode on which is a potential danger if in proximity to the ground or people.

SupaflyFPV, 2024-07-29:
you can use the failsafe setup for this.

stubb0rnCoder, 2024-07-29:
Do you mind explaining further? All I can see is setting the PRY channels to a static (hold or set) value which is not what I am proposing here.

SteveCEvans, 2024-07-29:
If you had to disarm manually that means you had control. In that case you could simply have reduced the throttle to minimum to achieve the desired effect.

closed 2024-08-21 by haslinghuis