betaflight/betaflight #13801: SITL not working Ubuntu 22.04 Gaz11

IRBaboon595, 2024-08-05:
### Describe the bug

I have walked through this instruction:
https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md
Installed gazebo from repo - got v.11.
I am able to see my remote's sticks in betaflight configurator (getting correct values for 8 channels).
Do not see ANYTHING in lo using tcpdump (ports 9002 or 9003). 
And no flight or reaction within Gazebo model. Ardupilot though works fine.

### To Reproduce

Used ubuntu 22.04.
Build BF master: make TARGET=SITL
Installed Gazebo: sudo apt install gazebo
As in the instruction:
for Ubunutu 16.04: cp /usr/share/gazebo-8/worlds/iris_arducopter_demo.world .

change real_time_update_rate in iris_arducopter_demo.world: <real_time_update_rate>0</real_time_update_rate> to <real_time_update_rate>100</real_time_update_rate> this suggest set to non-zero

Next - I run 4.6.0 sitl:
./obj/main/betaflight_SITL.elf 
Next - Gazebo:
gazebo --verbose ./iris_arducopter_demo.world
At last - run tcpdump:
sudo tcpdump -i lo udp and port 9002

### Expected behavior

Expect some kind of working as described [here](https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md) and [here](https://betaflight.com/docs/development/sitl). Or at least - udp datagramms in tcpdump.

### Support ID

```txt
2465d48d-9a0d-4fbd-a3b5-c80e8ee9deb3
```

### Flight controller

SITL

Swesser, 2024-09-04:
Hello! How have you connected your controller? is it a USB RC? 

I am struggling, MSP thing does not work for me...

IRBaboon595, 2024-09-09:
> Hello! How have you connected your controller? is it a USB RC?
> 
> I am struggling, MSP thing does not work for me...

Yes I have.
I have soldered a TBS nano to FTDI uart - usb pcb, connected the last to PC and made a software that translates a SBUS from com port to MSP UDP port. At last I have opened a RX receiver port in BC which resulted in a UDP port in local.

closed 2024-10-17 by github-actions[bot]