betaflight/betaflight #13802: HDZero OSD renders very slowly in Releases 4.5.0 and later

JDRotor, 2024-08-05:
### Describe the bug

Since the release of 4.5.0 the HDZero OSD is rendering very slowly.   At times a refresh of the OSD can take as long a 5 secs.  In addition, between rendering the  OSD element on the screen freeze.   Main issue is with critical telemetry data i.e. Receiver Battery status. 

### To Reproduce

Issue is a 100% operational issue.   When running 4.5.+ FW with HDZero 

### Expected behavior

Very Slow OSD rendering updating .

### Support ID

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0.KAACK_V12 May 17 2024 / 20:36:02 (cd18bed46) MSP API: 1.46
# config rev: 93dfc69

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HOBBYWING_XROTORF7CONV
manufacturer_id HOWI
mcu_id 003c00165746501520303431
signature 

# name: Five33_SB1

# resources
resource SERIAL_RX 5 NONE
resource PINIO 3 D02

# feature
feature -AIRMODE
feature TELEMETRY
feature OSD

# serial
serial 1 64 115200 57600 0 115200
serial 2 131073 115200 57600 0 115200

# beeper
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB

# aux
aux 0 0 0 1900 2075 0 0
aux 1 13 2 1975 2025 0 0
aux 2 28 2 1400 1600 0 0
aux 3 35 4 1450 1550 0 0
aux 4 40 3 1350 1650 0 0
aux 5 42 5 1050 2100 0 0
aux 6 49 4 1925 2075 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0
adjrange 1 0 1 900 2100 29 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
[... 314 lines cut ...]
set yaw_srate = 39
set throttle_limit_type = SCALE
set throttle_limit_percent = 70

# restore original rateprofile selection
rateprofile 0

# save configuration
```

### Flight controller

Hobbywing XROTORF7CONV & Aiken F7

### Other components

VTX is HDZero Race v3

Various ESCs Aikon and Hobbywing

### Add any other context about the problem that you think might be relevant here

Back up testing evidence available including Task dumps and rendering refresh timing study.   See excel file.

[BF 4.5.0 Test Protocal w. data 2.xlsx](https://github.com/user-attachments/files/16503598/BF.4.5.0.Test.Protocal.w.data.2.xlsx)

JDRotor, 2024-08-05:
[image]
[image]

JDRotor, 2024-08-06:
[image]
[image]
[image]
[image]

JDRotor, 2024-08-06:
[image]

JDRotor, 2024-08-06:
[image]
[image]

JDRotor, 2024-08-06:
Reopened - close by error

JDRotor, 2024-08-06:
YouTube video showing rendering times:

4.4.3
https://www.youtube.com/shorts/eNiq1gbx9as?feature=share

4.5.0
https://www.youtube.com/shorts/yDjbkiAEHZQ?feature=share

robhaswell, 2024-08-06:
I'm also experiencing this on this quad:

```
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0.KAACK_V12 Jun 21 2024 / 20:18:19 (cd18bed46) MSP API: 1.46
# config rev: 140b82d

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF7
manufacturer_id CLRA
mcu_id 001500184353500120383248
signature 

# name: OpenRacer Orange

# feature
feature TELEMETRY
feature OSD

# serial
serial 1 64 115200 57600 0 115200
serial 4 131073 115200 57600 0 115200

# led
led 0 0,0::CY:0
led 1 1,0::CY:0
led 2 2,0::CY:0
led 3 3,0::CY:0
led 4 4,0::CY:0
led 5 5,0::CY:0
led 6 6,0::CY:0
led 7 7,0::CY:0
led 8 8,0::CY:0
led 9 9,0::CY:0
led 10 10,0::CY:0
led 11 11,0::CY:0
led 12 12,0::CY:0
led 13 13,0::CY:0
led 14 14,0::CY:0
led 15 15,0::CY:0
led 16 0,1::CY:0
led 17 1,1::CY:0
led 18 2,1::CY:0
led 19 3,1::CY:0
led 20 4,1::CY:0
led 21 5,1::CY:0
led 22 6,1::CY:0
led 23 7,1::CY:0
led 24 8,1::CY:0
led 25 9,1::CY:0
led 26 10,1::CY:0
led 27 11,1::CY:0
led 28 12,1::CY:0
led 29 13,1::CY:0
led 30 14,1::CY:0
led 31 15,1::CY:0

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 26 0 1700 2100 0 0
aux 5 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# vtx
vtx 0 4 0 0 1 900 1300
vtx 1 4 0 0 2 1300 1700
vtx 2 4 0 0 3 1700 2100

# rxfail
rxfail 6 s 1500

# master
set gyro_lpf1_static_hz = 0
set gyro_to_use = FIRST
set dyn_notch_count = 1
set dyn_notch_q = 600
set dyn_notch_min_hz = 200
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = 95,-70,-77,1
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set motor_kv = 1800
set dshot_bidir = ON
set motor_output_reordering = 1,0,3,2,4,5,6,7
set align_board_yaw = 180
set vbat_max_cell_voltage = 440
set yaw_motors_reversed = ON
set rpm_limit = ON
set rpm_limit_value = 26000
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_vbat_pos = 385
set osd_rssi_pos = 353
set osd_link_quality_pos = 2082
set osd_rssi_dbm_pos = 377
set osd_tim_1_pos = 54
set osd_tim_2_pos = 3489
set osd_flymode_pos = 374
set osd_throttle_pos = 2614
set osd_vtx_channel_pos = 3112
set osd_crosshairs_pos = 280
set osd_craft_name_pos = 393
set osd_pilot_name_pos = 2549
set osd_warnings_pos = 14675
set osd_avg_cell_voltage_pos = 2475
set osd_disarmed_pos = 429
set osd_spec_logo_pos = 14401
set osd_sys_vtx_temp_pos = 2114
set osd_stat_bitmask = 5092
set osd_canvas_width = 50
set osd_canvas_height = 18
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_freq = 5732
set displayport_msp_fonts = 0,0,0,0
set rpm_filter_weights = 100,50,100
set rpm_filter_min_hz = 150
set rpm_filter_fade_range_hz = 100
set craft_name = OpenRacer Orange
set pilot_name = Mopatop

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 10
set iterm_relax_cutoff = 45
set iterm_limit = 500
set pidsum_limit_yaw = 1000
set throttle_boost = 2
set p_pitch = 41
set i_pitch = 86
set d_pitch = 41
set f_pitch = 137
set p_roll = 38
set i_roll = 78
set d_roll = 37
set f_roll = 125
set d_min_roll = 25
set d_min_pitch = 27
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 87
set feedforward_jitter_factor = 2
set feedforward_boost = 18
set feedforward_max_rate_limit = 100
set dyn_idle_min_rpm = 20
set simplified_pids_mode = RP
set simplified_i_gain = 115
set simplified_d_gain = 85
set simplified_pi_gain = 85
set simplified_dmax_gain = 140
set simplified_feedforward_gain = 105
set simplified_pitch_d_gain = 95
set simplified_pitch_pi_gain = 105
set tpa_rate = 70
set tpa_breakpoint = 1250
set ez_landing_limit = 5

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 30
set pitch_rc_rate = 30
set yaw_rc_rate = 30
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 53
set pitch_srate = 53
set yaw_srate = 53
set throttle_limit_percent = 80

rateprofile 1

# rateprofile 1
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 15
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 100

rateprofile 2

# rateprofile 2
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 15
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 53
set pitch_srate = 53
set yaw_srate = 53

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

JDRotor, 2024-08-06:
> I now have an HDZero setup and will investigate over the next couple of days. Thank you for the config files and videos.

Thanks Steve,

 

I’ve been working with Ivan for a while on this.   I do have a couple Blackbox files with Debug set for Timing Accuracy and Scheduler Deter., that I could not figure out how to load into the Issues Report.

If that would be of any help let me know as to the best way to send them

 

We do appreciate all you guy do for the hobby.   Thanks!

 

Regards,

Joe DiDia

SteveCEvans, 2024-08-07:
Taking a look at the relative timings. With 4.4.3, using a logic analyzer to generate a pulse every time the craft name is displayed (once per frame) we get:

[image]

With master we get the following which is a massive improvement.

[image]

SteveCEvans, 2024-08-07:
> Thanks Steve, I’ve been working with Ivan for a while on this. I do have a couple Blackbox files with Debug set for Timing Accuracy and Scheduler Deter., that I could not figure out how to load into [...]

If you create a zip archive you should be able to add that to a comment here. I'm still trying to recreate the issue. I'm seeing master redraw the display with great timing accuracy. I'll now check the 4.5 release as perhaps we've improved things already in master...

SteveCEvans, 2024-08-07:
Well, with 4.5.0 I get exactly the same as with master. A big improvement of 4.4.3.
[image]

SteveCEvans, 2024-08-07:
Hooked up an ESC and motors and now I'm seeing issues.

The below shows an armed period between the red markers. Clearly during that period the rate at which the OSD_STATE_TRANSFER state is invoked is reduced.

[image]

JDRotor, 2024-08-07:
BBL Log on Timing Acc and Scheduler Deter.
[Timing_Scheduler_BBL.zip](https://github.com/user-attachments/files/16534183/Timing_Scheduler_BBL.zip)

What we are experiencing is real - If its a settings issue, I'd love suggestions.   But when we back Flash below 4.5.0 the issue delay goes away (and we use the exact CLI dump (Diff All) in all tests.  Let me know what else you need.

SteveCEvans, 2024-08-07:
I get a significant improvement as shown below by reducing the value of `scheduler_relax_osd` from 25 to 1. Update rate then remains consistent. Could you please check if this fixes the issue for you. Note that this is a workaround not a fix, but I'd be interested in your observations.

[image]

JDRotor, 2024-08-07:
Ok I. An work on that in a couple of hours and will report back.  Regards,Joe DiDia

image.png (view on web)

JDRotor, 2024-08-07:
Steve, 
I took some quick measurements on one quad with HDZero setup, with 3 different variants:
#1) 4.5.0 with scheduler_relax_osd (set at default) = 25,  
#2) 4.5.0 with scheduler_relax_osd = 1 and 
#3) 4.4.3 with default scheduler_relax_osd =25    

The average screen rendering time observed via visual/stopwatch method:  
#1 = 2.50 sec; 
#2 = 1.30 sec;
#3 = 0.90 sec;

So changing the scheduler_relax_osd  value to 1 does yield a rendering time improvement of about 50% but still slower than Release v4.4.3

Let me know if you need more.  Thanks

SteveCEvans, 2024-08-09:
This appears to be HDZero specific. Using your config on a ZEEZF7V3 I ran a build of master and with the HDZero I saw the reported issues with the motors armed. Unplugging the HDZero with the system still armed and plugging in a Walksnail system the display then rendered correctly with the appropriate warnings flashing as they should. Unplugging the Walksnail and going back to the HDZero the display became laggy again.

[image]

I'll have to dig deeper into the timing of the MSP data-stream and the responses.

limonspb, 2024-08-10:
What HDzero FW version you running?
I'm in Hobbywing F7 FC, hdzero 1.5.0, don't see this issue.

JDRotor, 2024-08-10:
I'm on 1.6.0 and Mike is on 1.5.0.   All my quads are equipped with Hobbywing F7 and Mike's are a mix of Aikon F7 and Hobbywing F7.   We have issues on all 9 of our quads.   Did you test with motors armed and running?

SteveCEvans, 2024-08-10:
My current firmware version string is:

```
app:9.3.1 rx:75 va:191
```

Certainly doesn’t look like the versions you’re talking about, so am I looking in the wrong place?

JDRotor, 2024-08-10:
You have the latest - the numering is different when using the USB dongle to flash  - never figure out why.

limonspb, 2024-08-10:
This also could be interesting, there is a bad rate mismatch fix for hdzero:
https://github.com/hd-zero/hdzero-vtx/pull/213

But according to Hdzero developer that could cause initial slow screen updated, but not the  constant slow updates.

JDRotor, 2024-08-10:
I ran some stopwatch checks with just one quad loaded with 4.4.3 and 4.5.1 and then flashed the Race V3 VTx with v.
1.5.0 and 1.6.0

4.4.3 is still the benchmark with nearly instant OSD rendering no surprise.  The time reading for the two different HDZ vtx firmware were close but 1.5.0 was a bit better:

FW 1.5.0 avg rendering after arming was 1.30 seconds and 
FW 1.6.0 avg rendering after arming was 1.63 sec.   
4.4.3 benchmark was around 0.7 sec (but quick difficult to measure accurately with a stopwatch).

limonspb, 2024-08-10:
> I ran some stopwatch checks with just one quad loaded with 4.4.3 and 4.5.1 and then flashed the Race V3 VTx with v. 1.5.0 and 1.6.0
>
> 4.4.3 is still the benchmark with nearly instant OSD rendering no [...]

Could you please also try that baud rate fix from hdzero?
Here are the fw files to DL:
https://github.com/hd-zero/hdzero-vtx/actions/runs/10312835870

JDRotor, 2024-08-10:
> Could you please also try that baud rate fix from hdzero? Here are the fw files to DL: https://github.com/hd-zero/hdzero-vtx/actions/runs/10312835870

Ok I ran the "Stopwatch timings on the same Quad above but with the HDZ Vtx firmware with the "Baud Rate" fix (on only the v4.5.1 firmware.   The average rendering time after arming was very close or slightly slower than the previous test of the v 1.6.0 firmware 1.83 sec.    

My conclusion is it really did not improve the rendering time.

SteveCEvans, 2024-08-11:
I have a fix. The issue was that with the default (or increased) filtering in place there was insufficient time for the OSD task to run frequently enough. So I've split the drawing and rendering of a given object into two parts, scheduled one after the other. This has made it much easier to schedule the OSD task. Need to test that this doesn't break analog video, for example. I'll raise a PR shortly.

JDRotor, 2024-08-11:
Awesome!  Thank you for the update.

limonspb, 2024-08-11:
> I have a fix. The issue was that with the default (or increased) filtering in place there was insufficient time for the OSD task to run frequently enough. So I've split the drawing and rendering of [...]

Hmm wondering what's changed in 4.5 in terms of filters performance

JDRotor, 2024-08-11:
I tried to flash commit #13813 to test but the FW build failed - am I too earlier?

SteveCEvans, 2024-08-11:
Which FC do you need a build for?

JDRotor, 2024-08-11:
Hobbywing F7 XRotorConv

 

And if it works, I can have it tested on AikonF7 too but HW first. 

 

 

Regards,

Joe

SteveCEvans, 2024-08-11:
[betaflight_4.6.0_STM32F7X2_HOBBYWING_XROTORF7CONV.hex.zip](https://github.com/user-attachments/files/16575807/betaflight_4.6.0_STM32F7X2_HOBBYWING_XROTORF7CONV.hex.zip)

JDRotor, 2024-08-11:
I successfully flashed the Hex but the 10.10.0 does not support the 4.6.0 FW.   I'm looking for a Configurator nightly but unable to locate it.   Can you help me out please?  JD

SteveCEvans, 2024-08-11:
https://app.betaflight.com/

JDRotor, 2024-08-11:
Did bench stopwatch testing and all seems good.   The rendering was about 0.5 sec slower than our benchmark testing on 4.4.3 but acceptable and it looks to render updates reliably during arming and throttle punches, etc.  -  I will test more rigorously this week.  JD

SteveCEvans, 2024-08-11:
[betaflight_4.6.0_STM32F7X2_AIKONF7.hex.zip](https://github.com/user-attachments/files/16576052/betaflight_4.6.0_STM32F7X2_AIKONF7.hex.zip)

I'm battling with the unit tests now... once that's fixed you should be able to build yourself.

JDRotor, 2024-08-11:
Ok Thanks,  I'm good here for now.    I can wait at this point.   I have Bench tested two quad and all seems the same.   
Only odd thing that happened on the second quad was during the CLI backup loading, its errored on two commands "Set Canvas Height =" and "Set Canvas Width =".    Said "unrecognized" not sure why - I used the same cli list that I have been using.   In any case it ran ok and the OSD Canvas seemed fine.   Going to assume there's a syntax change in 4.6.0 that's in this code.   

I will test fly these Tuesday.  Thanks again!

SteveCEvans, 2024-08-11:
See https://github.com/betaflight/betaflight/pull/13595

JDRotor, 2024-08-11:
Got it – I guess I’m one of those guys messing around with the Canvas dimensions 😊    Good to know it’s no longer needed.   I  had issues with this in the very early days of HDZ set up (Dec 2022) and I never stopped putting it into my personal CLI presets .   

 

 

Regards,

Joe

JDRotor, 2024-08-12:
Steve,   IS #13813 still the PR number to use for flashing.   I kind of lost track of things.   Thanks

SteveCEvans, 2024-08-12:
Yes

SteveCEvans, 2024-08-13:
What ELRS rate are you using? I'm using 500Hz with 1:2 telemetry ratio. Faster will increase load of course.

JDRotor, 2024-08-13:
We are both using F1000 w/ Telem at 1:2.    Flew today with the load and the Rendering was definitely acceptable.  All elements stayed the on the displace through out the flight was acceptable refresh times.   Do you have any questions idea when we will be able to flash using the Configurator with online loading?Regards,Joe DiDia

SteveCEvans, 2024-08-13:
Thanks for testing.

JDRotor, 2024-08-14:
> Thanks for testing.

The thing is I'm not having much luck with flashing via the Configurator (either version 11.0 or 10.10) except 1 time from 10.10.   Is there a trick I'm not aware of.   I'd like to load it in different riggs w/ different FCs to test.

JDRotor, 2024-08-15:
Steve,
I assume you are done with the coding but wanted to let you know that we did more testing today, in the air with 7 quads, with different ESCs and FCs.   All had acceptable OSD rendering times.   So we concur with your fix.   Now we did see some OSD lagging when the ELRS telemetry ratio was turned down to 1:128 (I dont think that should be a surprise, the telemetry elements (i.e. voltage/RSSI dbm/Link Qty) would blank out (not be visible for 2 - 4 sec) but when the ELRS telem ratio was set on 1:2 the screen was always hot and refreshed quickly.    I hope all that is helpful.

closed 2024-08-22 by blckmn