betaflight/betaflight #13806: Changing the direction of rotation of the servo does not work on versions newer than 4.4.3

TheMonomah, 2024-08-08:
### Describe the bug

Expected behavior (how it works on firmware 4.4.3 and higher) - to invert the direction of rotation of the servo, you need to swap the min and max values ​​on the servo tab for the servo you want to change.
Malfunction (on firmware 4.5 and newer) - if you swap the min and max values, the servo completely stops working and responding to changes in the aux channel value, the servo always only perceives the value 2000.

### To Reproduce

In the servo tab, swap the min and max values, the servo will stop working, its channel value will always be equal to 2000

### Expected behavior

When the min and max values ​​are swapped, the servo starts to rotate in the opposite direction (when activated)

### Support ID

```txt
if necessary, I will copy from 5 different controllers
```

### Flight controller

The problem has been tested and identified on all flight controllers I have had - SpeedyBee F405 V4, Foxeer F405 V2, SoloGood F722, Iflight F435 (AT32)

### Other components

Bug checks were performed on a bare flight controller to eliminate the influence of other components

haslinghuis, 2024-08-08:
https://github.com/betaflight/betaflight/commit/413d05d6c48e5ac1aac92789feb40d02b2aa64ce

TheMonomah, 2024-08-08:
> [413d05d](https://github.com/betaflight/betaflight/commit/413d05d6c48e5ac1aac92789feb40d02b2aa64ce)

I see this code and understand that it extended the servo range, but does it also break the rotation inversion?

haslinghuis, 2024-08-09:
Can you test if the issue is resolved using this firmware:

[betaflight_4.6.0_STM32F405_SPEEDYBEEF405V4.zip](https://github.com/user-attachments/files/16563859/betaflight_4.6.0_STM32F405_SPEEDYBEEF405V4.zip)

TheMonomah, 2024-08-09:
> Can you test if the issue is resolved using this firmware:
> 
> [betaflight_4.6.0_STM32F405_SPEEDYBEEF405V4.zip](https://github.com/user-attachments/files/16563859/betaflight_4.6.0_STM32F405_SPEEDYBEEF405V4.zip)
> 

Hi, I flashed the firmware, but the configurator does not allow me to configure it, it says that 10.10 is too old a version for this firmware.

haslinghuis, 2024-08-10:
Check https://github.com/betaflight/betaflight-configurator?tab=readme-ov-file#unstable-testing-versions

TheMonomah, 2024-08-10:
> Check https://github.com/betaflight/betaflight-configurator?tab=readme-ov-file#unstable-testing-versions

On this firmware everything works great👍

haslinghuis, 2024-08-12:
@TheMonomah can you provide a support ID please.

TheMonomah, 2024-08-13:
> @TheMonomahМожете ли вы предоставить идентификатор поддержки, пожалуйста?

I updated the post and added the identifier d4b5f5e3-fa29-405a-a633-555d902a69c9

ledvinap, 2024-08-14:
From code, is does not seem that this worked in BF in 4.4.3

TheMonomah, 2024-08-14:
> From code, is does not seem that this worked in BF in 4.4.3

I don't know what is visible from the code, but in reality it works on 4.4.3 and I did it just yesterday on 4.4.3 and have done it before many times

ledvinap, 2024-08-14:
@TheMonomah : 
Can you post cli `diff` `version` (or support id, if supported) for working version?
 d4b5f5e3-fa29-405a-a633-555d902a69c9 has servo min=1000, max=2000

Setting min > max is bug in configurator (it is refused in CLI)

haslinghuis, 2024-09-16:
Stalled.

closed 2024-09-16 by haslinghuis

eugene771, 2025-01-10:
Hi all.
However the issue is relevant and still present.
It's absolutely necessary in some cases to have ability to set 2000 for minimum tumblet position and 1000 - for max.
Also MID setting should also better work properly for mid tumbler position (now it's not, using normal direction it is counted as MIN+MAX/2).
Thanks.

haslinghuis, 2025-01-10:
@eugene771 can you test #14140 on 4.6 development with custom define `SERVO_LEGACY`  ?