betaflight/betaflight #13812: Blackbox datapoints missing T-Motor F722 AIO 35A

Figure-itout, 2024-08-10:
### Describe the bug

Datapoints missing in blackbox logs on internal storage.
[image]

### To Reproduce

Just log data to blackbox and do a flight with throttle above 50%.

### Expected behavior

Blackbox logging is complete with all the data points.

### Support ID

```txt
954e4918-d6d7-4512-afd7-b5b3388ff032
```

### Flight controller

T-Motor F722 AIO 35A (TMOTORF7_AIO(STM32F7X2))

### Other components

ELRS EP1, version 3.3.2 (UART 1)
Walksnail HD Pro VTX (latest) with MSP for OSD (UART 2)

### How are the different components wired up (including port information)

ELRS EP1 -> UART 1
WS VTX -> UART 2 MPS+DSP (115200)

### Add any other context about the problem that you think might be relevant here

I tried with BF 4.3, 4.4 and 4.5 (and finally with 4.5.1).
I bought a second board (same model, different shop), no difference.
I put the board in another build, no different.

I think the SPI-bus is shared with the internal barometer:
baro_spi_device = 3
Allowed range: 0 - 5

flash_spi_bus = 3
Allowed range: 0 - 4

gyro_1_spibus = 1
Allowed range: 0 - 4

gyro_2_spibus = 0
Allowed range: 0 - 4

Devices detected: SPI:2, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6500, ACC=MPU6500, BARO=BMP280

I tried with the barometer disabled in Betaflight, no difference.
The dataflash is a Winbond 25Q128JVSQ 2031. I also flashed with custom defines USE_W25Q128JV and USE_W25Q128JV_QUADSPI, both no difference. The ID in the status message (FLASH: JEDEC ID=0x00ef4018 16M) is for a W25Q128FV, as you can find in src/main/drivers/flash/flash_w25q128fv.c.

All cables are as far away as possible from the dataflash.

Figure-itout, 2024-08-10:
Blackbox logfile
[btfl_all.bbl.zip](https://github.com/user-attachments/files/16570612/btfl_all.bbl.zip)

Figure-itout, 2024-08-10:
I also tried with lower PID Loop frequency and with lower loggin frequency. Disabled all options and no debug data, but still no complete data of the information that is logged.

ot0tot, 2024-08-28:
Yes, flash and baro share a SPI bus. Lower your PID loop to 4kHz and disable the baro on the Configuration tab and see if the blackbox logs properly.

SteveCEvans, 2024-09-02:
Gaps are typically caused by the FLASH memory not being able to keep up. That said, disabling the Baro is worth a try. Try logging at 1/2 or 1/4 rate. You rarely need full speed logs.

Figure-itout, 2024-09-08:
I tried with PID Loop at 4 Khz and sample rate to 1/2 and 1/4.
But both is still till 50% to 60% throttle logged. Above it is missing the datapoints.

closed 2024-10-16 by github-actions[bot]