betaflight/betaflight #13817: OSD battery voltage reading is sometimes wrong | BF 4.6.0-dev

rpisarew, 2024-08-12:
### Describe the bug

I have very often a wrong voltage reading in the OSD. It shows 0V or 2.14V, sometimes it blinks and switch between this two values. When this is happening I still get the right value over telemetry. My current workaround is to power cycle until it works, usually around 1-2 times.

### To Reproduce

- flash latest bf 4.6.0-dev
- plug in battery
- check voltage in OSD (this issue happens randomly)

### Expected behavior

right voltage in the OSD

### Support ID

```txt
fefdc32f-49af-42d5-8677-10e74c6ec625
```

### Flight controller

BETAFPV F4 1S 5A AIO (BETAFPV411)

### Add any other context about the problem that you think might be relevant here

Im currently on this [build](https://github.com/betaflight/betaflight/pull/13797)

haslinghuis, 2024-08-12:
Flash and OSD are sharing SPI bus. Does it help to disable blackbox?

rpisarew, 2024-08-13:
> Flash and OSD are sharing SPI bus. Does it help to disable blackbox?

It still happens, but I have the feeling that is more rarely.

haslinghuis, 2024-08-13:
Perhaps check with this change: https://github.com/betaflight/betaflight/pull/13813

rpisarew, 2024-08-13:
> Perhaps check with this change: #13813

still happening.

closed 2026-09-26 by blckmn