betaflight/betaflight #13819: F745 out of control at 8K

howels, 2024-08-14:
### Describe the bug

FLYWOO F745 AIO board with AM32 ESCs. Upon arming at 8K the craft jerks violently and motors can be heard rapidly changing speed. Stable flight is impossible at 8K.
Reverting to 4K produces smooth flight with no issues.

### To Reproduce

Set 8K PID loop

### Expected behavior

F745 should be able to run 8K PID loop 

### Support ID

```txt
Id: "f98ea6da-4919-4160-83da-6fa1fd4d47df"
```

### Flight controller

FLYWOO GOKU GN745 1.2 https://flywoo.net/products/goku-gn745-40a-aio-bl_32-mpu6000-25-5-x-25-5

### Other components

O3 air unit (disconnected for these tests)
ExpressLRS 3.4 @500Hz
1505 motors
150mm 3.5" frame

### Add any other context about the problem that you think might be relevant here

[betaflight_logs.zip](https://github.com/user-attachments/files/16611310/betaflight_logs.zip)

First log with SCHEDULER_DETERMINISM, second log with SECHEDULER.

SteveCEvans, 2024-08-14:
Can you please provide the output of the `tasks` command to give the IDs of the tasks to help interpret the logs.

Could you also please provide a log using `TIMING_ACCURACY` and also `TASKS` with `scheduler_debug_task` set to the ID (from the `tasks` output) of the OSD task.

howels, 2024-08-14:
> Also, what ESC firmware/version are you using? The webpage says it's BLHeli32, but have you got AM32? I know what the first line of your issue states, but what to check, and which version.

AM32 2.12 has been flashed to the board after purchase. It's a F051 target with the same configuration as iflight.

howels, 2024-08-14:
Log for TASKS with task ID set to 20 for the OSD task,
[TASKS_btfl_all.bbl.zip](https://github.com/user-attachments/files/16612049/TASKS_btfl_all.bbl.zip)

howels, 2024-08-14:
By comparison at 4K PID loop there are no issues performing flights in full acro mode. Log of full flight at 4K PID.
[4K_btfl_002.bbl.zip](https://github.com/user-attachments/files/16614155/4K_btfl_002.bbl.zip)

SteveCEvans, 2024-08-22:
You should have `set dshot_bitbang = ON`. That doesn't fix the issue, but timer based DSHOT makes it worse and messes up the gyro timing.

Testing with the above config, albeit with bit banged DSHOT enabled and the setup below, I'm seeing significant arming/disarming delays. I'll keep investigating.

[image]

SteveCEvans, 2024-08-23:
This appears to be related to AM32.

I've compared the `FLYWOOF745`  to a `IFLIGHT_F745_AIO` which uses Bluejay.

Using an identical configuration, they both sit at ~60% load, but when armed the load decreases to about 56% on the `IFLIGHT_F745_AIO`.

[image]

But on the `FLYWOOF745` it increases to 76%.

[image]

I presume this is time spent waiting for telemetry data. Looking into that now.

SteveCEvans, 2024-08-23:
Just tested again using an ELRS rx rather than MSP RC, and I'm get very different results, so I'm going to put the above down to the USB serial driver which is known to really mess up timings.

Using ELRS at 500Hz I'm seeing no delays whatsoever. The differences between my config and that posted above are `dshot_bitbang = ON` and the following serial port config.

```
serial 1 64 115200 57600 0 115200
serial 2 131073 115200 57600 0 115200
```

[image]

SteveCEvans, 2024-08-23:
@howels Please test again with `dshot_bitbang = ON`. Without that I see a loop time of 127us rather than the correct 125us.

howels, 2024-09-12:
> @howels Please test again with `dshot_bitbang = ON`. Without that I see a loop time of 127us rather than the correct 125us.

Still have this on my list but other things have got in the way, should be able to do the tests next week.

haslinghuis, 2026-10-03:
Resolved by using 4K pid loop

closed 2026-10-03 by haslinghuis