betaflight/betaflight #13821: GOKU F722 PRO Mini V2 Flight Controller baro Barometer: DP310 not recognize on betaflight

titeuf007, 2024-08-14:
### Describe the bug

hey everybody
sorry for my english
i just to install the latest beta on my goku f722 and everything go well until i saw that my baro is not recognize
could you say me if it s an common issues with this dp310 baro?

### To Reproduce

i have installed my firmware twice to be sure it s not a firmware failure

### Expected behavior

need you help

### Support ID

```txt
kkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkkk
```

### Flight controller

GOKU F722 PRO Mini V2

flywoo, 2024-08-15:
Hello, we tested that the barometer works normally in BF4.5.1. Maybe the barometer is faulty and not recognized. You can contact technical support by email at flywoofpv@outlook.com

closed 2024-08-16 by github-actions[bot]

marcbogadi, 2024-09-22:
I had the same problem, issue was caused by wrong firmware. i had FLYWHOOF7DUAL target on the FC. After flashing the right target FYLWHOOF722PRO the baro works. For the Mini the target is FYLWHOOF722PROV2

saga111a, 2025-01-19:
@marcbogadi I had same problem, no baro recognition in Flywoo Goku 722 pro v2 mini. I tried change i2c, remove gps, change BF versions etc and there was no effects. Your advise helped me. I think it shoud be **wrong auto detection flash target in BF for Flywoo Goku 722 pro v2 mini!**

haslinghuis, 2025-01-29:
Solved on discord