betaflight/betaflight #13823: Fix open redirect on website, being used in phishing campaigns

cgite, 2024-08-15:
### Is your feature request related to a problem? Please describe

The URL https://analytics.betaflight.com/redir?dest= is actively being used in phishing campaigns.

### Describe the solution you'd like

Remove open redirect

### Describe alternatives you've considered

Blocking domain

nerdCopter, 2024-08-21:
@VitroidFPV @freasy  ?

freasy, 2024-08-21:
@cgite can you show me examples?

@blckmn

nerdCopter, 2024-08-21:
jsut my useless 2 cents...
- _maybe_ a node_module is the source, as grepping redirect shows many results.
- mitigation might include disabling redirects or whitelisting target domain (betaflight.com only)
- seemingly mitigation is a feature of the server. unsure if using apache httpd, asp.net or another. (check if url is local, allow redirect, else redirect home)